slowtec/tokio-modbus #1: rtu non-functional when used

kallisti5, 2018-02-21:
**main.rs**
```rust
extern crate futures;
extern crate tokio_core;
extern crate tokio_modbus;
extern crate tokio_serial;
extern crate tokio_service;
use tokio_modbus::*;

pub fn main() {
    use tokio_core::reactor::Core;
    use futures::future::Future;
    use tokio_serial::{BaudRate, Serial, SerialPortSettings};

    let mut core = Core::new().unwrap();
    let handle = core.handle();
    let tty_path = "/dev/ttyUSB0";
    let server_addr = 0x01;

    let mut settings = SerialPortSettings::default();
    settings.baud_rate = BaudRate::Baud19200;
    let mut port = Serial::from_path(tty_path, &settings, &handle)
        .expect(&format!("Unable to open serial device '{}'", tty_path));
    port.set_exclusive(false)
        .expect("Unable to set serial port exlusive");

    let task = Client::connect_rtu(port, server_addr, &handle).and_then(|client| {
        println!("Reading a sensor value");
        client
            .read_holding_registers(0x082B, 2)
            .and_then(move |res| {
                println!("Sensor value is: {:?}", res);
                Ok(())
            })
    });

    core.run(task).unwrap();
}
```

***Cargo.toml***
```toml
[dependencies]
futures = "*"
tokio = "*"
tokio-core = "*"
tokio-serial = "*"
tokio-service = "*"
tokio-modbus = { version = "*", default-features = false, features = ["rtu"] }
```

***Results***
```shell
error[E0599]: no function or associated item named `connect_rtu` found for type `tokio_modbus::Client` in the current scope
  --> src/main.rs:25:16
   |
25 |     let task = Client::connect_rtu(port, server_addr, &handle).and_then(|client| {
   |                ^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error
```

flosse, 2018-02-21:
It looks like you are using `tokio_modbus` `v0.1.0` that has a [different API](https://docs.rs/tokio-modbus/0.1.0/tokio_modbus/struct.RtuClient.html).
The current master branch should lead to `v0.2.0` but first these issues should be closed:
- https://github.com/tokio-rs/tokio-proto/issues/204
- https://github.com/tokio-rs/tokio-proto/issues/202
- https://github.com/tokio-rs/tokio/issues/118

So in your case you can either use the `v0.1.0` API ([example](https://github.com/slowtec/tokio-modbus/blob/v0.1.0/examples/rtu-client.rs)) or point the `tokio-modbus` dependency to the git repo:

```toml
[dependencies.tokio-modbus]
version = "*"
git = "https://github.com/slowtec/tokio-modbus"
default-features = false
features = ["rtu"]
```

flosse, 2018-03-30:
I just published `v0.2.0` so you can use it as usual :)

closed 2018-03-30 by flosse