slowtec/tokio-modbus #3: The example does not run on Windows 10.

aleksey-r, 2018-11-30:
**main.rs**
```rust
extern crate futures;
extern crate tokio_core;
extern crate tokio_modbus;
extern crate tokio_serial;

use tokio_core::reactor::Core;
use futures::future::Future;
use tokio_serial::{Serial, SerialPortSettings};
use tokio_modbus::*;

pub fn main() {
    let mut core = Core::new().unwrap();
    let handle = core.handle();
    let port_path = "COM1";
    let server_addr = 0x01;

    let mut settings = SerialPortSettings::default();
    settings.baud_rate = 19200;
    let mut port = Serial::from_path(port_path, &settings, &handle).unwrap();
    port.set_exclusive(false).unwrap();

    let task = Client::connect_rtu(port, server_addr, &handle).and_then(|client| {
        println!("Reading a sensor value");
        client
            .read_holding_registers(0x082B, 2)
            .and_then(move |res| {
                println!("Sensor value is: {:?}", res);
                Ok(())
            })
    });

    core.run(task).unwrap();
}
```

**Cargo.toml**

```rust
[dependencies]  
futures = "*"  
tokio = "*"  
tokio-core = "*"
tokio-serial = "*"
tokio-service = "*"
tokio-modbus = { version = "*", default-features = false, features = ["rtu"] }
```

**Results**
```rust
C:/Users/test/.cargo/bin/cargo.exe run --package modbus_test --bin modbus_test
   Compiling tokio-modbus v0.2.1
error[E0432]: unresolved import `tokio_serial::Serial`
 --> C:\Users\test\.cargo\registry\src\github.com-1ecc6299db9ec823\tokio-modbus-0.2.1\src\service\rtu.rs:3:5
  |
3 | use tokio_serial::Serial;
  |     ^^^^^^^^^^^^^^^^^^^^ no `Serial` in the root

error[E0432]: unresolved imports `tokio_serial::Serial`, `tokio_serial::SerialPortSettings`
 --> C:\Users\test\.cargo\registry\src\github.com-1ecc6299db9ec823\tokio-modbus-0.2.1\src\client.rs:9:20
  |
9 | use tokio_serial::{Serial, SerialPortSettings};
  |                    ^^^^^^  ^^^^^^^^^^^^^^^^^^ no `SerialPortSettings` in the root
  |                    |
  |                    no `Serial` in the root

error[E0433]: failed to resolve. Could not find `tcp` in `proto`
  --> C:\Users\test\.cargo\registry\src\github.com-1ecc6299db9ec823\tokio-modbus-0.2.1\src\server.rs:75:39
   |
75 |                 TcpServer::new(proto::tcp::Proto, addr)
   |                                       ^^^ Could not find `tcp` in `proto`

error[E0277]: the size for values of type `dyn futures::Future<Item=frame::Response, Error=std::io::Error>` cannot be known at compilation time
  --> C:\Users\test\.cargo\registry\src\github.com-1ecc6299db9ec823\tokio-modbus-0.2.1\src\service\rtu.rs:52:9
   |
52 |         Box::new(result)
   |         ^^^^^^^^ doesn't have a size known at compile-time
   |
   = help: the trait `std::marker::Sized` is not implemented for `dyn futures::Future<Item=frame::Response, Error=std::io::Error>`
   = note: to learn more, visit <https://doc.rust-lang.org/book/second-edition/ch19-04-advanced-types.html#dynamically-sized-types-and-the-sized-trait>
   = note: required by `<std::boxed::Box<T>>::new`

error[E0277]: the size for values of type `dyn futures::Future<Item=frame::Response, Error=std::io::Error>` cannot be known at compilation time
  --> C:\Users\test\.cargo\registry\src\github.com-1ecc6299db9ec823\tokio-modbus-0.2.1\src\service\rtu.rs:40:13
   |
40 |         let result = self.service.call(req).and_then(move |resp| {
   |             ^^^^^^ doesn't have a size known at compile-time
   |
   = help: the trait `std::marker::Sized` is not implemented for `dyn futures::Future<Item=frame::Response, Error=std::io::Error>`
   = note: to learn more, visit <https://doc.rust-lang.org/book/second-edition/ch19-04-advanced-types.html#dynamically-sized-types-and-the-sized-trait>
   = note: all local variables must have a statically known size
   = help: unsized locals are gated as an unstable feature

error: aborting due to 5 previous errors

Some errors occurred: E0277, E0432, E0433.
For more information about an error, try `rustc --explain E0277`.
error: Could not compile `tokio-modbus`.

To learn more, run the command again with --verbose.

Process finished with exit code 101

```

flosse, 2018-11-30:
@aleksey-r Thanks for reporting!
I added a configuration appveyor to run a windows CI:
https://ci.appveyor.com/project/flosse/tokio-modbus

I'll have a closer look at it soon.

flosse, 2018-11-30:
@aleksey-r Could you try to use `tokio-serial = "0.6"` and remove the `default-features = false`?

```diff
diff --git a/Cargo.toml b/Cargo.toml
index 6b12a17..54f75c6 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -9,4 +9,4 @@ tokio = "*"
 tokio-core = "*"
 tokio-serial = "*"
 tokio-service = "*"
-tokio-modbus = { version = "*", default-features = false, features = ["rtu"] }
+tokio-modbus = { version = "*", features = ["rtu"] }
diff --git a/src/main.rs b/src/main.rs
index 12b22f4..a449e98 100644
--- a/src/main.rs
+++ b/src/main.rs
@@ -5,7 +5,7 @@ extern crate tokio_serial;
 
 use tokio_core::reactor::Core;
 use futures::future::Future;
-use tokio_serial::{Serial, SerialPortSettings};
+use tokio_serial::{Serial, BaudRate, SerialPortSettings};
 use tokio_modbus::*;
 
 pub fn main() {
@@ -15,7 +15,7 @@ pub fn main() {
     let server_addr = 0x01;
 
     let mut settings = SerialPortSettings::default();
-    settings.baud_rate = 19200;
+    settings.baud_rate = BaudRate::Baud19200;
     let mut port = Serial::from_path(port_path, &settings, &handle).unwrap();
     port.set_exclusive(false).unwrap();
```

flosse, 2018-11-30:
@aleksey-r Could you try to reproduce this with this repo: https://github.com/flosse/tokio-modbus-issue-3

aleksey-r, 2018-11-30:
```
lex@DEN D:\
$ git clone https://github.com/flosse/tokio-modbus-issue-3.git
Cloning into 'tokio-modbus-issue-3'...
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 7 (delta 0), reused 7 (delta 0), pack-reused 0
Unpacking objects: 100% (7/7), done.

lex@DEN D:\
$ cd tokio-modbus-issue-3

lex@DEN D:\tokio-modbus-issue-3
$ cargo build
   Compiling semver-parser v0.7.0
   Compiling winapi-build v0.1.1
   Compiling winapi v0.3.6
   Compiling rand_core v0.3.0
   Compiling void v1.0.2
   Compiling libc v0.2.44
   Compiling stable_deref_trait v1.1.1
   Compiling winapi v0.2.8
   Compiling nodrop v0.1.13
   Compiling cfg-if v0.1.6
   Compiling byteorder v1.2.7
   Compiling memoffset v0.2.1
   Compiling lazy_static v1.2.0
   Compiling scopeguard v0.3.3
   Compiling slab v0.4.1
   Compiling futures v0.1.25
   Compiling lazycell v1.2.0
   Compiling rustc-serialize v0.3.24
   Compiling scoped-tls v0.1.2
   Compiling slab v0.3.0
   Compiling take v0.1.0
   Compiling smallvec v0.2.1
   Compiling unreachable v1.0.0
   Compiling owning_ref v0.4.0
   Compiling crossbeam-utils v0.6.1
   Compiling log v0.4.6
   Compiling arrayvec v0.4.8
   Compiling ws2_32-sys v0.2.1
   Compiling kernel32-sys v0.2.2
   Compiling ktmw32-sys v0.1.1
   Compiling advapi32-sys v0.2.0
   Compiling rand_core v0.2.2
   Compiling rand_hc v0.1.0
   Compiling rand_isaac v0.1.1
   Compiling rand_xorshift v0.1.0
   Compiling semver v0.9.0
   Compiling smallvec v0.6.7
   Compiling lock_api v0.1.5
   Compiling log v0.3.9
   Compiling crossbeam-epoch v0.6.1
   Compiling rustc_version v0.2.3
   Compiling num_cpus v1.8.0
   Compiling rand_chacha v0.1.0
   Compiling rand_pcg v0.1.1
   Compiling parking_lot_core v0.3.1
   Compiling rand v0.6.1
   Compiling crossbeam-deque v0.6.2
   Compiling tokio-executor v0.1.5
   Compiling tokio-service v0.1.0
   Compiling tokio-timer v0.2.8
   Compiling tokio-current-thread v0.1.4
   Compiling rand v0.5.5
   Compiling net2 v0.2.33
   Compiling rand v0.4.3
   Compiling rand v0.3.22
   Compiling tokio-threadpool v0.1.9
   Compiling parking_lot v0.6.4
   Compiling iovec v0.1.2
   Compiling miow v0.2.1
   Compiling winreg v0.4.0
   Compiling bytes v0.4.11
   Compiling serialport v1.0.1
   Compiling tokio-io v0.1.10
   Compiling mio v0.6.16
   Compiling tokio-codec v0.1.1
   Compiling tokio-fs v0.1.4
   Compiling tokio-reactor v0.1.7
   Compiling mio-serial v0.6.0
   Compiling tokio-udp v0.1.3
   Compiling tokio-tcp v0.1.2
   Compiling tokio v0.1.13
   Compiling tokio-core v0.1.17
   Compiling tokio-proto v0.1.1
   Compiling tokio-serial v0.6.0
   Compiling tokio-modbus v0.2.2 (https://github.com/slowtec/tokio-modbus?branch=v0.2.x#e63ef477)
error[E0432]: unresolved import `tokio_serial::Serial`
 --> C:\Users\lex\.cargo\git\checkouts\tokio-modbus-4bf0058fcb35ae92\e63ef47\src\service\rtu.rs:3:5
  |
3 | use tokio_serial::Serial;
  |     ^^^^^^^^^^^^^^^^^^^^ no `Serial` in the root

error[E0432]: unresolved import `tokio_serial::Serial`
 --> C:\Users\lex\.cargo\git\checkouts\tokio-modbus-4bf0058fcb35ae92\e63ef47\src\client.rs:9:20
  |
9 | use tokio_serial::{Serial, SerialPortSettings};
  |                    ^^^^^^ no `Serial` in the root

error[E0432]: unresolved import `tokio_serial::SerialPortSettings`
 --> C:\Users\lex\.cargo\git\checkouts\tokio-modbus-4bf0058fcb35ae92\e63ef47\src\client.rs:9:28
  |
9 | use tokio_serial::{Serial, SerialPortSettings};
  |                            ^^^^^^^^^^^^^^^^^^ no `SerialPortSettings` in the root

error[E0277]: the size for values of type `dyn futures::Future<Error=std::io::Error, Item=frame::Response>` cannot be known at compilation time
  --> C:\Users\lex\.cargo\git\checkouts\tokio-modbus-4bf0058fcb35ae92\e63ef47\src\service\rtu.rs:52:9
   |
52 |         Box::new(result)
   |         ^^^^^^^^ doesn't have a size known at compile-time
   |
   = help: the trait `std::marker::Sized` is not implemented for `dyn futures::Future<Error=std::io::Error, Item=frame::Response>`
   = note: to learn more, visit <https://doc.rust-lang.org/book/second-edition/ch19-04-advanced-types.html#dynamically-sized-types-and-sized>
   = note: required by `<std::boxed::Box<T>>::new`

error[E0277]: the size for values of type `dyn futures::Future<Error=std::io::Error, Item=frame::Response>` cannot be known at compilation time
  --> C:\Users\lex\.cargo\git\checkouts\tokio-modbus-4bf0058fcb35ae92\e63ef47\src\service\rtu.rs:40:13
   |
40 |         let result = self.service.call(req).and_then(move |resp| {
   |             ^^^^^^ doesn't have a size known at compile-time
   |
   = help: the trait `std::marker::Sized` is not implemented for `dyn futures::Future<Error=std::io::Error, Item=frame::Response>`
   = note: to learn more, visit <https://doc.rust-lang.org/book/second-edition/ch19-04-advanced-types.html#dynamically-sized-types-and-sized>
   = note: all local variables must have a statically known size

error: aborting due to 5 previous errors

Some errors occurred: E0277, E0432.
For more information about an error, try `rustc --explain E0277`.
error: Could not compile `tokio-modbus`.

To learn more, run the command again with --verbose.
```

flosse, 2018-11-30:
hm...that is interesting.

flosse, 2018-11-30:
ok, I think tokio-serial is not available for windows in v0.6: https://github.com/berkowski/tokio-serial/blob/33bf3e57f05afa1642a3025d225b7128e35af3b7/src/lib.rs#L27

flosse, 2018-11-30:
So could you try to run the newest [tokio-serial example](https://github.com/berkowski/tokio-serial/blob/master/examples/serial_println.rs) on windows?

flosse, 2018-12-01:
@aleksey-r I meant the latest example of `tokio-serial` only: https://github.com/berkowski/tokio-serial/blob/master/examples/serial_println.rs

flosse, 2018-12-01:
...anyway, increasing the version of the `tokio-serial` dependency would be a breaking change.

@uklotzde what do you think, shall we release a `v0.3.x` that is equal `v0.2` but depends on `tokio-serial` v3.1?

uklotzde, 2018-12-01:
I will rebase my current _work-in-progress_ on _master_ and adopt a new version number.

aleksey-r, 2018-12-01:
> 
> 
> @aleksey-r I meant the latest example of `tokio-serial` only: https://github.com/berkowski/tokio-serial/blob/master/examples/serial_println.rs

tokio-serial example works fine on windows 10

> @aleksey-r could you please checkout the master of https://github.com/flosse/tokio-modbus-issue-3 and try it again?

```
error[E0599]: no method named `set_exclusive` found for type `tokio_serial::Serial` in the current scope
  --> src\main.rs:20:10
   |
20 |     port.set_exclusive(false).unwrap();
   |          ^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: Could not compile `tokio-modbus-issue-3`.

To learn more, run the command again with --verbose.
```

flosse, 2018-12-01:
The current master [is now working on appveyor](https://ci.appveyor.com/project/flosse/tokio-modbus/build/job/1mucv6fn12hw05vq) :)

@aleksey-r 
Could you retry it? (`cargo update && cargo run`)?

aleksey-r, 2018-12-01:
```
lex@DEN D:\tokio-modbus-issue-3
$ cargo update
    Updating crates.io index
    Updating git repository `https://github.com/slowtec/tokio-modbus`
    Updating tokio-modbus v0.2.2 (https://github.com/slowtec/tokio-modbus#e4ff598b) -> #aed9fd3a

lex@DEN D:\tokio-modbus-issue-3
$ cargo run
   Compiling tokio-modbus v0.2.2 (https://github.com/slowtec/tokio-modbus#aed9fd3a)
   Compiling tokio-modbus-issue-3 v0.1.0 (D:\tokio-modbus-issue-3)
error[E0599]: no method named `set_exclusive` found for type `tokio_serial::Serial` in the current scope
  --> src\main.rs:20:10
   |
20 |     port.set_exclusive(false).unwrap();
   |          ^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: Could not compile `tokio-modbus-issue-3`.

To learn more, run the command again with --verbose.
```

aleksey-r, 2018-12-01:
Works fine after removing the set_exclusive method.

flosse, 2018-12-01:
So I think we can close this issue now :)
@uklotzde are there any other things that should be included in `v0.3`?

closed 2018-12-01 by flosse