slowtec/tokio-modbus #8: Change the device address after connecting.

aleksey-r, 2018-12-04:
Is there a way to change the server address after establishing a connection with the device (connect_rtu)?

flosse, 2018-12-04:
> Is there a way to change the address after establishing a connection with the device (connect_rtu)?

I don't think so.

@uklotzde Could this be a small feature for the 0.3.x release?

uklotzde, 2018-12-04:
:+1: for including this extension in the next release.

**Proposal**
Clients could be assigned an immutable default `device_id: u8` (TCP: unit_id, RTU: address) upon construction. Both client traits will be extended by duplicating all methods with an additional _device_id_ parameter that allows to conditionally override the client's default _device_id_ for some calls.

uklotzde, 2018-12-09:
See also: https://github.com/slowtec/tokio-modbus/issues/9#issuecomment-445523963

uklotzde, 2019-01-11:
@flosse Please add a milestone for 0.3.0 and assign it to this issue.

closed 2019-04-04 by uklotzde