slowtec/tokio-modbus #9: Support for unit ids

axelfaure, 2018-12-04:
First of all thank you for sharing this crate with us!

I guess this is a feature request but it may just be a documentation issue.
I need to setup a modbus TCP communication relying on unit identifiers. I think this is not supported by this crate but could you please confirm it ?

If the feature is actually missing, is there any plan to implement it ?

uklotzde, 2018-12-05:
As proposed [here](https://github.com/slowtec/tokio-modbus/issues/8#issuecomment-444105443) this can be implemented by introducing a `device_id: u8` for both TCP and RTU transports. What do you think?

uklotzde, 2018-12-05:
Correct, the unit identifier is currently hard-coded for TCP.

uklotzde, 2018-12-06:
Another variant of sending messages to selected slave devices (address range: 1-247) is sending broadcast messages to all connected devices by using the address 0.

uklotzde, 2018-12-07:
We also need to consider the use case of using one of the reserved addresses 248-255, e.g. for devices with a fixed broadcasting address 253.

uklotzde, 2018-12-09:
Heavy work in progress: https://github.com/uklotzde/tokio-modbus/tree/devices

With `tcp::connect_device()` you will be able to connect to a specific _device_ aka _unit_ instead of using a direct connection (default : device/unit_id = 0xFF) to the coupler. It will also be possible to switch dynamically between devices at runtime for both RTU and TCP connections, see the _rtu-client.rs_ example.

@flosse I'm still not sure about the API. Let's talk tomorrow.

uklotzde, 2019-01-11:
@flosse Please add a milestone for 0.3.0 and assign it to this issue.

axelfaure, 2019-02-12:
Thank you for the work you put in this new version!

I may have missed something but I think sync::Context should implement SlaveContext as well

closed 2019-04-04 by uklotzde