slowtec/tokio-modbus #18: Move from tokio-core and tokio-proto to tokio

cbourjau, 2018-12-17:
First and foremost! Thanks a lot for sharing this library! I was trying to use it today and noticed that it uses the deprecated `tokio-core` and `tokio-proto` crates. Is there any particular issue holding back the migration to the current `tokio` crate?

flosse, 2018-12-17:
@cbourjau Your suggested migration is totally in our minds :)
Unfortunately there is no drop-in replacement of `tokio-proto` so we have to refactor the current implementation first. There is an ongoing progress in #10 :)

BTW:
Contributions are very welcome ;-)

flosse, 2019-01-10:
According to https://github.com/tokio-rs/tokio/issues/118#issuecomment-452969665 https://github.com/tower-rs/tokio-tower seems to be the replacement for `tokio-proto` but it seems to be buggy: https://github.com/tower-rs/tokio-tower/issues/2

uklotzde, 2019-04-04:
tokio-proto complicates handling of timeouts as discussed in #12

flosse, 2020-02-04:
now there is no more tokio-proto in the master :)

closed 2020-02-04 by flosse