slowtec/tokio-modbus #19: Custom TCP connect timeout ?

jvelo, 2019-02-20:
Hello,

Is it possible to set a custom TCP connect timeout ?
I'm new to rust, I might be missing some things obvious in the way tokio works

If it's not possible and of interest, I'm willing to give it a go.

Thanks!

flosse, 2019-02-28:
@jvelo Sorry for the late reply.
The TCP connection currently is managed by the deprecated `tokio-proto` crate that no longer is maintained. Unfortunately there is no drop-in replacement for that.
I already ask for a possible replacement more than a year ago (https://github.com/tokio-rs/tokio-proto/issues/202#issue-292208479) but there is not really a progress on that :(
This is why we're going to remove the dependency entirely. The corresponding issue is discussed in #18 
Probably we first will release a `v0.3.x` (#10 ) and will work on a `tokio-proto` replacement in a `v0.4.x` release.
Until then you might use a custom wrapper with a timeout feature.

MightyPork, 2020-04-27:
any progress / plans for this?

it made me ditch the sync client and write this async abomination instead (my first time using futures, too)

```rust
let mut rt = tokio::runtime::Runtime::new().unwrap();
rt.block_on(async {
        let mut ctx = tokio::time::timeout(Duration::from_secs(3),
                                           tokio_modbus::client::tcp::connect_slave(socket_addr, Slave(12)))
            .await.unwrap().unwrap();

        // more stuff
    });
```

glennpierce, 2021-05-11:
> any progress / plans for this?
>
> it made me ditch the sync client and write this async abomination instead (my first time using futures, too)
>
> ```rust
> let mut rt = [...]

This saved me. Ugly but works

flosse, 2024-09-12:
I consider the issue to be resolved.
Feel free to open it again if it is not.

closed 2024-09-12 by flosse