slowtec/tokio-modbus #30: Having trouble running a simple read register example

glennpierce, 2019-06-04:
Hi I was wondering if you can see why I can't get the following example to work.
I get the error
ExceptionResponse { function: 3, exception: GatewayTargetDevice } 

```
pub fn main() {

    let mut core = Core::new().unwrap();
    let handle = core.handle();
    let socket_addr = "10.0.128.181:502".parse().unwrap();

    let task = tcp::connect(&handle, socket_addr).and_then(|ctx| {

        ctx
            .read_holding_registers(0x16, 2)
            .and_then(move |data| {
                println!("Response is '{:?}'", data);
                Ok(())
            })
    });
    core.run(task).unwrap();
}
```

I know the ip code is correct as the following Python code works

```
c = ModbusClient()
c.host("10.0.128.181")
c.port(502)
c.unit_id(1)
c.debug(False)
c.open()

regs = c.read_holding_registers(0x16, 2)

if regs:
    print(regs)
    lsb = regs[0]
    msb = regs[1]
    print (msb << 16) | lsb
else:
    print("read error")

c.close()
```

What could be the issue here ?
Thanks

uklotzde, 2019-06-04:
Maybe you need to switch the unit/slave id of the client context to 1 like you did in your Python code?

```
ctx.set_slave(Slave::from(1))
```
See also: [`trait SlaveContext`](https://docs.rs/tokio-modbus/0.3.4/tokio_modbus/slave/trait.SlaveContext.html)

This is needed if the TCP device acts as a gateway. Otherwise the reserved unit id = 0xFF is used as the default for directly accessing the TCP device.

uklotzde, 2019-06-04:
Once the slave/unit id is set it is used for all subsequent requests.

glennpierce, 2019-06-04:
Thanks that was it.
I didn't replicate that bit from Python as it was not necessary to have it for the Python code. It must default to slave id 1.

Thanks

uklotzde, 2019-06-04:
Feedback is always welcome.

closed 2019-06-04 by uklotzde

dave-andersen, 2021-10-02:
It would be really handy to have this documented in the TCP example, even if just a comment:
// If the device acts as a gateway, set the unit ID to 1:
// ctx.set_slave(Slave::from(1))

Took me quite a while to figure out why my Rust version wasn't working when porting a client from Go's goburrow/modbus, which also sets the unit ID as the Python library does and didn't need this configuration.