slowtec/tokio-modbus #32: Error inspection is difficult because error types are not exported

laumann, 2019-07-16:
There are some errors that could be survived, for instance if an illegal address is used. But some errors, like a broken connection, are hard if not impossible to survive.

However, all the error types in the library seem to be `pub(crate)` meaning I cannot downcast any errors to eg `ExceptionResponse` (even though it implements `Error`).

I would suggest making `ExceptionResponse` and `Exception` ([here](https://github.com/slowtec/tokio-modbus/blob/master/src/frame/mod.rs)) public to allow anyone handling errors when using the libray to inspect the cause of the error. As it stands, I imagine I have to call `.description()` and match on the returned string.

I haven't thought the implications through, I just wanted to file this request and am hoping to open a discussion from here.

flosse, 2019-07-16:
@laumann Thanks for your feedback!
I'll chat on that with @uklotzde soon.

laumann, 2019-07-16:
Thank you @flosse I'm up for opening a PR, I could maybe outline what I'd like to be able to do as a user?

flosse, 2019-07-16:
@laumann we had a busy day today and were not able to handle your issue. I just skimmed the code and as far as I can see your proposal makes sense.

> Thank you @flosse I'm up for opening a PR, I could maybe outline what I'd like to be able to do as a user?

That would be awesome! And i guess we can easily ship a `v0.3.5` without braking changes :)

laumann, 2019-07-17:
> @laumann we had a busy day today and were not able to handle your issue. I just skimmed the code and as far as I can see your proposal makes sense.

Me too :smile: Thanks for taking a look.

> > Thank you @flosse I'm up for opening a PR, I could maybe outline what I'd like to be able to do as a user?
> 
> That would be awesome! And i guess we can easily ship a `v0.3.5` without braking changes :)

Great, I'll submit a PR soon enough and then we can take it from there.

flosse, 2020-03-07:
Solved in aedb60c

closed 2020-03-07 by flosse

laumann, 2020-03-08:
Thanks! That looked easy enough :-)