slowtec/tokio-modbus #33: Unable to Close TCP Client Connection

DBTaylor, 2019-09-13:
Target: ARMv7 (musl)

I have a warp webserver that upon a request will open a connection and read some registers, and I want the connection to close after the registers are successfully read. Using the async API, I expected the TCP client connections to close when contexts are dropped, but they do not (netstat shows them in established state). I tried Context::disconnect, but that causes a panic ('internal error: entered unreachable code'). What is the correct way to disconnect the TCP connection?

```rust
let read_input_registers = get2()
        .and(path!("rir" / u16 / u16)
        .and(path::end())
        .and_then(move |start, count|{
            let (p, c) = futures::sync::oneshot::channel::<String>();
            remote.spawn(move |handle|
                client::tcp::connect(handle, localhost)
                .and_then(move |ctx| ctx.read_input_registers(start, count)/*.and_then(move |res| ctx.disconnect().then(|_| Ok(res)))*/)
                .then(|res|{
                    p.send(match res{
                        Ok(data) => format!("{:?}", data),
                        Err(err) =>  format!("{:?}", err)
                    }).unwrap_or(());
                    Ok(())
                })
            );
            c.map_err(|err| warp::reject::custom(err))
        }));
```
I've switched to the modbus crate until I can figure this out. I'd prefer to use this one since it's non-blocking.

flosse, 2019-09-14:
@DBTaylor Thanks for your feedback.
We'll try to reproduce this soon.
Probably this issue is related to the outdated `tokio-proto` crate (see #18).
The next version of tokio-modbus will definitely get rid of it and will use the new async/await features of Rust. So if you're interested to help feel free to contact us :)

flosse, 2019-09-14:
@uklotzde If you're bored: can you have a look at this?

DBTaylor, 2019-09-14:
@flosse Thanks for the quick response. Monday I'll put together a minimal program that reproduces this - I'll also check it on a Tier 1 target to make sure it's not a target specific issue. Glad to hear async/await support is coming - way easier to read than future combinators. I would totally help if I had any idea what I was doing.

DBTaylor, 2019-09-16:
@flosse here is a reproduction: https://github.com/DBTaylor/modbus-tcp-leak

It appears to be target agnostic. This issue sounds related, but it appears to have been fixed:
https://github.com/tokio-rs/tokio-proto/issues/194

flosse, 2019-09-16:
@DBTaylor Thanks for the example repo. I can reproduce the behavior on my machine.

I tried to build it with the current `tokio-proto` master but unfortunately it's still the same :(

uklotzde, 2019-09-16:
I've prepared a quick fix:

https://github.com/slowtec/tokio-modbus/pull/34

```
[patch.crates-io]
tokio-modbus = { git = "https://github.com/uklotzde/tokio-modbus.git", branch = "tcp_disconnect" }
```

After invoking disconnect() the connections will enter TIME_WAIT and should disappear after ~30 seconds (depending on your config). Does this solve your issues?

DBTaylor, 2019-09-16:
@uklotzde yep that works!

uklotzde, 2019-09-16:
Great, I will prepare a release.

Your example with the testing instructions was a great help to quickly solve this bug!

PR #34 (2019-09-17): Implement disconnect() for TCP clients
First attempt to fix #33.

The TCP implementation for the `disconnect()` request that drops connections by inserting a *poison pill* was missing. We needed this to get rid of exclusive connections to the serial port that got stale sometimes.

Unfortunately, TCP connections are not closed immediately but remain in TIME_WAIT for some time, before they disappear.
closed 2019-09-17 by uklotzde