slowtec/tokio-modbus #37: Modbus TCP / RTU Gateway

hanusek, 2020-01-10:
Hello. 
I need simple example with modbus gateway.
Please help me.

flosse, 2020-01-10:
> Please help me.

Could you provide a bit more information?
Do you have a repository with your current implementation?
Where exactly do you need help?

hanusek, 2020-01-10:
> > Please help me.
> 
> Could you provide a bit more information?
> Do you have a repository with your current implementation?
> Where exactly do you need help?

Hello, Thx for reply.
My repo: https://github.com/hanusek/ModbusGateway

flosse, 2020-01-14:
> My repo: https://github.com/hanusek/ModbusGateway

hm.... I just receive an 404 :-\

hanusek, 2020-01-15:
> > My repo: https://github.com/hanusek/ModbusGateway
> 
> hm.... I just receive an 404 :-\

OK. I'm fixed it. Check now.

flosse, 2020-01-18:
I cannot see the usage of `tokio-modbus` in your code.
Please checkout the [examples](https://github.com/slowtec/tokio-modbus/blob/master/examples/) first, try to adjust them for your usecase and precisely describe your problem or question.

flosse, 2020-01-18:
I'm going to close this issue until it's clear what's exactly the question/problem. Feel free to reopen it. If you need commercial support you're welcome to contact us.

closed 2020-01-18 by flosse