slowtec/tokio-modbus #38: no method named `read_holding_registers` found for type `impl std::future::Future`

conorf50, 2020-01-22:
Firstly, thanks to both Slowtec for creating this library and Omnioiot for updating it with ```async/await``` support!
My code below is based on omnioiot's ```async``` branch (#35 ) so I could use the new async/await functionality.

I'm running into an issue when trying to use the ```tokio_modbus::client::sync::tcp::connect``` function. It seems to be returning the result of the connection from my ```connect_sync()``` function wrapped in a ```Future```  even though the function signature states that it only returns a Result

The [source](https://github.com/omnioiot/tokio-modbus/blob/1b9a36415c71c229b2d0af776a7e3d77d7f87d10/src/client/sync/tcp.rs#L8) for this function does not mention a Future. I may be missing something here but I'm confused as to why a sync method needs to be using a Future

Here is my code, the error is on line 8

Thanks for the help!

```
use tokio_modbus::prelude::*;
use tokio_modbus::client::tcp;
use tokio_modbus::client::sync::*; // TODO, refactor this to use async I/O

pub fn main() {

    let ctx = connect_sync();
    let power = ctx.read_holding_registers(40204, 1);
    println!("Power = {:?}", power);
    
}

async fn connect_sync() -> Result<tokio_modbus::client::sync::Context, std::io::Error> {
    
    let socket_addr = "127.0.0.1:5020".parse().unwrap();
    
    println!("attempting to connect");
    println!("{:?}", socket_addr);
    
    let ctx = tokio_modbus::client::sync::tcp::connect(socket_addr)?;
    Ok(ctx)
}
```

flosse, 2020-01-22:
@conorf50 Thank you for reporting.
I'm busy at the moment but can have a look at this at the end of next week.
Meanwhile you could ask @uklotzde, @Darksonn or @not-fl3

keisrk, 2020-01-22:
Hi guys, I'm also the one looking forward to asyn/await.  Thank you for this nice project @flosse . @conorf50 , could you try below?
```rust
use tokio_modbus::prelude::*;
use tokio_modbus::client::tcp;
use tokio_modbus::client::sync::*; // TODO, refactor this to use async I/O

pub fn main() {

    let ctx: Result<Context, _> = connect_sync();
    let mut ctx = ctx.unwrap();
    let power = ctx.read_holding_registers(40204, 1);
    println!("Power = {:?}", power);
    
}

/*async*/ fn connect_sync() -> Result<tokio_modbus::client::sync::Context, std::io::Error> {
    
    let socket_addr = "127.0.0.1:5020".parse().unwrap();
    
    println!("attempting to connect");
    println!("{:?}", socket_addr);
    
    let ctx = tokio_modbus::client::sync::tcp::connect(socket_addr)?;
    Ok(ctx)
}
```

Darksonn, 2020-01-23:
You defined `connect_sync` as an async fn, and all async fns automatically return a future (this is what it means to mark a function async). Thus the error regarding an `impl Future`. If you remove the async marker on `connect_sync`, it works:

```rust
use tokio_modbus::client::sync::Reader;

pub fn main() {

    let mut ctx = connect_sync().unwrap();
    let power = ctx.read_holding_registers(40204, 1);
    println!("Power = {:?}", power);
}

fn connect_sync() -> Result<tokio_modbus::client::sync::Context, std::io::Error> {

    let socket_addr = "127.0.0.1:5020".parse().unwrap();

    println!("attempting to connect");
    println!("{:?}", socket_addr);

    let ctx = tokio_modbus::client::sync::tcp::connect(socket_addr)?;
    Ok(ctx)
}
```

conorf50, 2020-01-23:
> You defined `connect_sync` as an async fn, and all async fns automatically return a future (this is what it means to mark a function async). Thus the error regarding an `impl Future`. If you remove the async marker on `connect_sync`, it works:

I had it marked as async because of a previous attempt at using async/await. I didn't realise that was what was causing the problem.

Thanks @Darksonn for the help!

flosse, 2020-02-04:
@conorf50 can we close this issue?

conorf50, 2020-02-04:
> @conorf50 can we close this issue?

Yes go ahead. Thanks everyone for the help

closed 2020-02-04 by flosse