slowtec/tokio-modbus #40: Support for function 43 / 14 Read Device Identification

psztoch, 2020-01-31:


flosse, 2020-03-07:
@psztoch Would you mind to implement this?

ptasz3k, 2020-04-19:
I can try it. I'm learning rust, using modbus professionally and probably will implement that function in my c and c# modbus libs. I think that it will be good exercise.

Any hints where should i start?

flosse, 2020-04-20:
> I can try it.

awesome!

> Any hints where should i start?

You can find an example described on page 43 in the [Modbus Application Protocol Specification (PDF)](http://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf).

[image]

So here are some possible steps to implement it:

1. Define the Request & Response
2. Write test(s) with the given example
3. Implement the processing algorithm
4. Extend the public API

gustavowd, 2023-04-04:
Hi.
Any chance to merge the read device identification function to the code?
Its stall since 2020.

XavierJEsteve, 2025-01-07:
Hello,
I will begin investigating this as I learn the package. Motivated to do so as I am new to ICS  and learning Rust. I will post questions here.

PR #328 (2025-09-16): Add Read device identification (Issue #40)
Initially, I intended to simply rebase #58 onto the current main branch. However, due to the changes in the library over time, it turned out to be more practical to rewrite the code—using @ptasz3k’s original work as a foundation/inspiration.

Thanks in advance for the review and feedback.

Closes #40.
commit 0b2afd54 (2025-09-16): Add Read device identification (Issue #40) (#328)

Closes #40.
closed 2025-09-16 by uklotzde