slowtec/tokio-modbus #43: Making a new release?

OrangeTux, 2020-02-03:
Thanks for your work on this project! 

The last release of this projects is `v0.3.5` which was released December 2019. Since than quite some changes have been made. Are you planning on releasing those soon?

flosse, 2020-02-04:
> Are you planning on releasing those soon?

Absolutely! I think we'll make it happen within the next few weeks.
You can already test or use it by adding 

```toml
[patch.crates-io]
tokio-modbus = { git = "https://github.com/slowtec/tokio-modbus" }
```
to your `Cargo.toml`.

flosse, 2020-02-04:
I created an separate [tracking issue for the v0.4.0 release](#47).

closed 2020-02-04 by flosse

OrangeTux, 2020-02-06:
Cool. Thanks!

flosse, 2020-03-08:
@OrangeTux FYI: The only missing [task before publishing v0.4.0](https://github.com/slowtec/tokio-modbus/issues/47#issue-559861229) is a manual RTU test with real hardware.