slowtec/tokio-modbus #44: Reponse to WriteSingleCoil is malformed

thosaa, 2020-02-04:
The response to WriteSingleCoil is not correct. 

I was testing with the Node.js module 'modbus-serial' as a TCP client and this crate as the Modbus TCP server, when I got an error on writeCoil (aka WriteSingleCoil). 

I then traced the packets using WireShark and it confirmes that Data is missing from the Response. 

```
Modbus
    .000 0101 = Function Code: Write Single Coil (5)
    [Request Frame: 2612]
    [Time from request: 0.000282788 seconds]
    Reference Number: 20
    Data: <MISSING>
[Malformed Packet: Modbus]
    [Expert Info (Error/Malformed): Malformed Packet (Exception occurred)]
        [Malformed Packet (Exception occurred)]
        [Severity level: Error]
        [Group: Malformed]

```
I am not sure on the validity of this page https://www.modbustools.com/modbus.html#function05, but it specifies that the response must be an echo of the request. 

I have clone the project and will try to fix it. I would like here, however, if you agreed with my findings or I'm wrong.

Regards,
Thomas

thosaa, 2020-02-04:
I think I have sorted out now. I am not able to test it, however.
I have checked the master branch out, made the changes and it builds. 
When I try to reference to it from my own project it states:
```
error[E0433]: failed to resolve: could not find `Server` in `tcp`
  --> src/main.rs:22:14
   |
22 |         tcp::Server::new(socket_addr).serve(|| Ok(modbus::MbServer));
   |              ^^^^^^ could not find `Server` in `tcp`
```
This is the dependency in Cargo.toml:
```
tokio-modbus = { version = "0.3.5", default-features = false, features = ["tcp"] }
```
This is the path override just below:
```
[patch.crates-io]
tokio-modbus = { path = "../tokio-modbus" }
```

I would like to test before making a pull request. Any help is appreciated, thanks.

/Thomas

flosse, 2020-02-04:
@thosaa thanks for reporting this bug and fixing it.

thosaa, 2020-02-04:
I cannot see what has changed since tag v0.3.5 to now, since I cannot run my code with the newest version:
I have now set my dep to:
```
tokio-modbus = { version = "0.3.5", default-features = false, features = ["tcp", "server"] }
```
and the same patch.crates-io as before. 

I now get this error:
```
23 |         tcp::Server::new(socket_addr).serve(|| Ok(modbus::MbServer));
   |                                       ^^^^^ the trait `tokio_modbus::server::service::Service` is not implemented for `modbus::MbServer`
```

And this is the way I call it:
```
    let _server = thread::spawn(move || {
        tcp::Server::new(socket_addr).serve(|| Ok(modbus::MbServer));
    });
```

flosse, 2020-02-04:
bug fixed with 8284940bae40ead5410f0abbe9942f0be9db2239

closed 2020-02-04 by flosse

flosse, 2020-02-04:
@thosaa I'll have a look at your questions at the weekend.