slowtec/tokio-modbus #47: Release v0.4.0

flosse, 2020-02-04:
# ToDo's:
- [ ] Update CHANGELOG
- [ ] Create a migration guide
- [ ] Test TCP client with real hardware ([ur20](https://github.com/slowtec/ur20-modbus))
- [ ] Test RTU client with real hardware ([SMT100](https://github.com/slowtec/truebner-smt100))
- [ ] Publish on crates.io

flosse, 2020-03-07:
@not-fl3 Can we merge `server/tcp.rs` and `server/tcp_server.rs`? For the `v0.4.0` it will be hidden by the `tcp-server-unstable` feature but for a `v0.4.1` we could release an implementation.

not-fl3, 2020-03-08:
> @not-fl3 Can we merge `server/tcp.rs` and `server/tcp_server.rs`? For the `v0.4.0` it will be hidden by the `tcp-server-unstable` feature but for a `v0.4.1` we could release an implementation.

totally agree, having very similar `Server` and `TcpServer` structs was very confusing. 
An attempt to fix it: https://github.com/slowtec/tokio-modbus/pull/56

closed 2020-03-13 by flosse

OrangeTux, 2020-03-16:
Thanks for releasing v0.4.0!