slowtec/tokio-modbus #50: Communicate with multiple modbus addresses concurrently

david-mcgillicuddy-moixa, 2020-02-18:
Hi, the modbus rtu example demonstrates how to use `set_slave` to talk to different addresses in turn. I would like to be able to talk to two different addresses simultaneously - send a request to device 1, send a request to device 2, then collect the two responses (possibly out of order).
Since only one packet will be sent on the wire at a time this certainly seems like it _should_ be possible to me.
However currently the first suspended future will hold onto the Rc's mutable borrow and the second one panics.
I tried also sharing the underlying serial with different `SharedContext`s but due to the `'static` requirement that isn't great either - I don't think that a mutex would avoid the problem that `Rc` had.

Will I need to manually implement packet routing on the incoming modbus frames? Surely there's something like this for TCP already?

I'd appreciate any pointers for how to go about this. I'm currently just experimenting with the unreleased 0.4.0 on master.

david-mcgillicuddy-moixa, 2020-02-19:
I realise that this was the kind of thing that TCP was invented for but unfortunately that's not an option for me. Since once a response comes in from a given address I'm going to be passing that in a message back upstream I see one possible interface look something like:
```
struct OutgoingRequest {
    address: u8,
    function_code: u8, // or an enum
    data: Vec<u8>, // or a slice
    callback: Box<dyn FnOnce(ModbusResponse) + 'static> // or a Future
}
```
Which would then register the callback and when a response came in, it would dispatch the response to a waiting callback (and then drop it).

david-mcgillicuddy-moixa, 2020-02-21:
After further reading of the spec this seems simply not supported at all by modbus over RS-285 sadly:
https://i.imgur.com/fkzYm40.png

Does this agree with your reading? I hope I am just misunderstanding something

zzeroo, 2020-05-25:
@david-mcgillicuddy-moixa You are right. In RTU is the happy path Request -> Waiting for reply -> Process reply. As shown in your image from the official standard. There is no option to send 2 Requests without waiting for the response. So all your ideas are not possible. This affects only the RTU standard, in TCP it should/ is possible!

I think this issue should be closed.

flosse, 2020-05-25:
> I think this issue should be closed.

@zzeroo thanks for clarifying!

closed 2020-05-25 by flosse