slowtec/tokio-modbus #51: Sync RTU Client

hanusek, 2020-02-18:
Hello. I need sync rtu client. Where I find?

zzeroo, 2020-02-19:
https://docs.rs/tokio-modbus/0.3.5/tokio_modbus/client/sync/rtu/fn.connect.html 

There is a `sync` module https://docs.rs/tokio-modbus/0.3.5/tokio_modbus/client/sync/index.html

hanusek, 2020-02-19:
> https://docs.rs/tokio-modbus/0.3.5/tokio_modbus/client/sync/rtu
> 
> There is a `sync` module https://docs.rs/tokio-modbus/0.3.5/tokio_modbus/client/sync/index.html

> ### The requested resource does not exist

david-mcgillicuddy-moixa, 2020-02-19:
Those both exist for me. There's also the direct example: https://github.com/slowtec/tokio-modbus/blob/master/examples/rtu-client.rs

hanusek, 2020-02-19:
> Those both exist for me. There's also the direct example: https://github.com/slowtec/tokio-modbus/blob/master/examples/rtu-client.rs

its asnyc.

hanusek, 2020-02-19:
I'm wrote a example.
```
		let mut settings = tokio_serial::SerialPortSettings::default();
		settings.baud_rate = 115200;
		println!("settings is '{:?}'", settings);

		const SLAVE_1: Slave = Slave(0x01);
		let result = sync::rtu::connect_slave("/dev/ttyUSB0", &settings, SLAVE_1);
		
		let mut ctx = match result
   		{  
       		Ok(ctx) => ctx,  
       		Err(error) => {  
				println!("There was a problem connect to slave : {:?}", error);
				return;  
		},
		};
		  

		let time_ms = time::Duration::from_millis(500);
		thread::sleep(time_ms);

		println!("Write");

		match ctx.write_single_register(0, 1)
		{
			Err(e) => {			
				println!("Error : '{:?}'", e);
			}
			Ok(f) => f,
		};
		thread::sleep(time_ms);

		match ctx.write_single_register(1, 2)
		{
			Err(e) => {			
				println!("Error : '{:?}'", e);
			}
			Ok(f) => f,
		};
		println!("Write done");

		let mut value = 1;
		loop
		{
			println!("Write");
			value = value + 1;
			match ctx.write_single_register(0, value)
			{
				Err(e) => {			
					println!("Error : '{:?}'", e);
				}
				Ok(f) => f,
			};
			
			thread::sleep(time_ms);

			println!("READ");
			let buff = ctx.read_holding_registers(0,2).unwrap();
			println!("Response is '{:?}'", buff);
		}
```
I have a problem when not receive response.
How set timeout?

david-mcgillicuddy-moixa, 2020-02-19:
I was not ever able to get timeouts working due to https://github.com/slowtec/tokio-modbus/issues/19
So I currently use the async code and race it with a tokio timeout using either `select2` in old tokio or `select!` in new tokio (depending on if you're using 0.3.5 or master of tokio-modbus).

hanusek, 2020-02-20:
> I was not ever able to get timeouts in _sync_ working due to #19
> So I currently use the async code and race it with a tokio timeout using either `select2` in old tokio or `select!` in new tokio (depending on if you're using 0.3.5 or master of tokio-modbus).

;(
I'm switching to https://github.com/zzeroo/libmodbus-rs

flosse, 2020-03-08:
> I'm switching to https://github.com/zzeroo/libmodbus-rs

Ok, so I'm closing this issue. Feel free to reopen it if there are questions left.

closed 2020-03-08 by flosse