slowtec/tokio-modbus #53: Trying to understand asynchronous client.

glennpierce, 2020-02-21:
Hi I am having trouble understanding how or the best way to make many requests in a multi threaded way. 
Basically I was trying to simulate talking to a couple of thousand modbus devices as quickly as possible. As I don't have those devices :) I was running your tcp-server.rs example with added delays of 60-80 ms. This seems reasonable as a simulation.

My trouble is developing the client code.

I want all the requests to be asynchronous. Once they all complete I would store these results into a db.

Would you recommend running code like

```
let socket_addr = "192.168.0.222:502".parse().unwrap();
let mut ctx = sync::tcp::connect(socket_addr).unwrap();
let buff = ctx.read_input_registers(0x1000, 7).unwrap();
```

in many threads and having each result send back through a channel ? or is there a better way built in using tokio async ?

Thanks

glennpierce, 2020-02-26:
I also believe there is an issue with sync::tcp::connect not closing the socket correctly.

I have the following code that eventually fails with too many open files error.
I have tested this with modbus-rs and it works.

```
extern crate chrono;
extern crate crossbeam_channel;

use std::sync::{Arc, Mutex};
use tokio_core::reactor::Core;
use futures::Future;
use tokio_modbus::prelude::*;
use systemd::daemon;

use chrono::{DateTime, NaiveDate, TimeZone, Utc};
use std::thread;

#[derive(Clone, Debug)]
struct MeterResult {
    name: String,
    ip: String,
    result: Option<u32>,
    last_dt: Option<DateTime<Utc>>
}

type MeterValue = (MeterResult, u32);

const NUMBER_OF_THREADS: usize = 10;
const NUMBER_OF_METERS: usize = 1000;

pub fn main() {
    let mut meters: Vec<MeterResult> = vec![];

    for i in 1..NUMBER_OF_METERS {
        let meter: MeterResult = MeterResult {name: format!("bi{}", i),
                                              ip: "127.0.0.1:5502".to_string(),
                                              result: None,
                                              last_dt: None
                                             };

        meters.push(meter);
    }
let mut values = Arc::new(Mutex::new(vec![]));
    let mut chunks = meters.chunks(NUMBER_OF_THREADS);

    for chunk in chunks {

        let mut threads = vec![];

        for meter in chunk {
            threads.push(thread::spawn( {

                let clone = Arc::clone(&values);
                let m = meter.clone();
                    move || {

                        let socket_addr = "127.0.0.1:5502".parse().unwrap();
                        let mut ctx = sync::tcp::connect(socket_addr).unwrap();
                        //let data = ctx.read_holding_registers(0x16, 2).unwrap();
                        //let value : u32 = ((data[1] as u32) << 16) | (data[0] as u32);
                        let mut v = clone.lock().unwrap();
                        v.push((m.clone(), 2));
                        //ctx.disconnect();
        //                drop(ctx);
        //                drop(socket_addr);
                    }
                }));
        }

        for t in threads {
            t.join().unwrap();
        }
        println!("batch done");
    }
}

```

DrSloth, 2021-01-31:
For everyone encountering this issue:
I once experienced a similar problem with the sync client. If you use tokio driven libraries you are actually best off to use them with async/await. One solution for the first question of this issue, is to use tokio::spawn to spawn tasks and at the end join them using the futures crate. This solution lets tokio decide how many threads to use and to concurrently run multiple tasks on one thread if for instance one of the slaves takes long to respond to the request.

aleksuss, 2021-12-03:
I can't figure out how to make `async` requests via one `Context`. The problem is `read_*` requests expect mutable reference to `Context`.  I'd like to write something like that:

```rust
    let mut context = tokio_modbus::client::tcp::connect("127.0.0.1:502".parse()?).await?;
    let (a, b) = tokio::join!(
        context.read_holding_registers(1, 1),
        context.read_holding_registers(2, 1),
    );
    
    context.disconnect().await?;
    dbg!(a, b);
```

I've tried to create a context for every request but a server doesn't like this. I thought about to wrap context in `Arc<RwLock>` but I think it's a not good idea.

DrSloth, 2021-12-03:
Because pretty much everything is uses `&mut self` you can just use `Arc<Mutex<Context>>`, i would recommend the [tokio Mutex](https://docs.rs/tokio/latest/tokio/sync/struct.Mutex.html).

 Internal state is mutated when making any networking request, reading from a `std::net::TcpStream` also requires it to be mutable for instance. It is generally not possible to make concurrent/parallel requests over the same connection. 

For concurrent writes to the same server you would need two contexts but many devices can't handle parallel requests correctly.

Hope that helps.

flosse, 2024-09-12:
I consider the issue to be resolved.
Feel free to open it again if it is not.

closed 2024-09-12 by flosse

thalesmaoanz, 2025-01-24:
If you allow me, I would like to join this discussion.

Currently, I communicate with 20 devices using Modbus RTU and Python. I’m planning to migrate to tokio-modbus, aiming to implement asynchronous communication similar to what was discussed in #50. However, since broadcast messages and their replies must be synchronized across all nodes in the Modbus network, I need to ensure that I wait for a reply from one device before proceeding to the next request.

At the moment, I handle this manually in my Python code. The issue arises when a device is powered off—it causes a timeout. During that timeout period, the program simply waits for the reply, leaving other parts of my code idle. If I can process other tasks during this time and avoid sending conflicting requests to other devices, that would address my concern.

I’ve reviewed https://github.com/slowtec/tokio-modbus/issues/59#issuecomment-1511428386, but I’m not entirely sure if it addresses what I need. Any guidance or suggestions would be greatly appreciated!

I've built this example to illustrate.

Every device has its own struct to store its own values and make its own request.
IA pointed that I can lock the access. This is what I currently do in python. But I can't fully understand if tokio will continue until reaches another bus request and wait it. With 500ms of timeout request, I can take 10s to read one device if other 19 are powered off. This is exactly my problem today.

Sorry for not being clear.

```rust
use tokio::sync::Mutex;
use tokio::task;
use tokio::time::{self, Duration, timeout};
use tokio_modbus::prelude::*;
use tokio_modbus::client::tcp;
use std::sync::Arc;

// Struct representing a Modbus device
struct Device {
    id: u8,
    address: String,
    port: u16,
}

impl Device {
    fn new(id: u8, address: String, port: u16) -> Self {
        Self { id, address, port }
    }

    async fn poll_data(
        &self,
        bus_lock: Arc<Mutex<()>>,
    ) -> Result<(), Box<dyn std::error::Error>> {
        let socket_addr = format!("{}:{}", self.address, self.port).parse()?;
        let mut ctx = tcp::connect(socket_addr).await?;

        let interval = Duration::from_secs(1); // Poll every second
        let timeout_duration = Duration::from_millis(500); // Max 500ms per request
        let mut interval_timer = time::interval(interval);

        loop {
            interval_timer.tick().await;

            // Acquire the lock to ensure exclusive access to the Modbus bus
            let _lock = bus_lock.lock().await;

            // Perform the Modbus request with a timeout
            let result = timeout(timeout_duration, ctx.read_input_registers(0, 10)).await;

            match result {
                Ok(Ok(response)) => {
                    println!("Device {} - Registers: {:?}", self.id, response);
                }
                Ok(Err(err)) => {
                    eprintln!("Device {} - Modbus error: {:?}", self.id, err);
                }
                Err(_) => {
                    eprintln!("Device {} - Request timed out after {:?}", self.id, timeout_duration);
                }
            }
            // Lock is automatically released here when `_lock` goes out of scope
        }
    }
}

#[tokio::main]
async fn main() -> Result<(), Box<dyn std::error::Error>> {
    // Create a shared mutex to synchronize access to the Modbus bus
    let bus_lock = Arc::new(Mutex::new(()));

    // Create a list of Modbus devices
    let devices = vec![
        Device::new(1, "192.168.0.101".to_string(), 502),
        Device::new(2, "192.168.0.102".to_string(), 502),
        Device::new(3, "192.168.0.103".to_string(), 502),
    ];

    // Spawn tasks for each device
    let mut tasks = vec![];
    for device in devices {
        let bus_lock = Arc::clone(&bus_lock); // Clone the lock for each task
        tasks.push(task::spawn(async move {
            if let Err(err) = device.poll_data(bus_lock).await {
                eprintln!("Error with device {}: {:?}", device.id, err);
            }
        }));
    }

    // Wait for all tasks to complete (in this case, they run indefinitely)
    for t in tasks {
        t.await?;
    }

    Ok(())
}

```