slowtec/tokio-modbus #54: TCP_NODELAY

klmurphy72, 2020-02-29:
A small piece of the Modbus_Messaging_Implementation_Guide_V1_0b is listed below.  I don't see anywhere in the code that is implemented.  Should this be added?  Is there some technical reason it is not?

TCP-NODELAY:
Small packets (called tinygrams) are normally not a problem on LANs, since most LANs
are not congested, but these tinygrams can lead to congestion on wide area networks.
A simple solution, called the "NAGLE algorithm", is to collect small amounts of data and
sends them in a single segment when TCP acknowledgments of previous packets
arrive.
In order to have better real-time behavior it is recommended to send small amounts of
data directly without trying to gather them in a single segment. That is why it is
recommended to force the TCP-NODELAY option that disables the "NAGLE algorithm"
on client and server connections.

flosse, 2020-03-07:
@klmurphy72 Thanks for posting this issue.

We can use [TcpStream::set_nodelay](https://tokio-rs.github.io/tokio-core/tokio_core/net/struct.TcpStream.html#method.set_nodelay)

Would you enable this by default? Or would you allow the user to decide the behavior? And would you mind to add a PR?

ptasz3k, 2020-04-20:
We are enabling it by default and it's not a value that is changeable in our c# library. The packets are sent physically as soon as they are sent by software. We never had any problems with that for five years already.