slowtec/tokio-modbus #57: Multiple servers on same tokio runtime

madprogrammer, 2020-04-16:
Hello,

I see this library is being actively developed, and I want to suggest to make it possible to spawn MODBUS TCP slave on the same tokio runtime (reactor) as other servers. I want to make an APP which listens on HTTP, MODBUS TCP and Serial port at the same time, and not to use threads for this purpose. Currently it seems there are not enough abstraction for the TCP server and its startup can't be easily customized. Perhaps you can suggest a way of doing what I need without modifying the `tokio-modbus` library?

flosse, 2020-04-17:
I think it should be possible but I haven't tried it yet.
We could take your usecase to add it to [examples](/slowtec/tokio-modbus/tree/master/examples) but I won't make it within the next days/weeks.

Would you mind to create draft a PR so I can guide you?

madprogrammer, 2020-04-17:
So far I got it to work as follows. Not sure how to get rid of `thread::spawn`. I'm just starting to play with Rust, and my original implementation is in Python with Twisted, where it all works well. I just like that Rust produces a single binary which is ready to run on my target ARM-based system, and it's kind of an overhead to deploy all Python stuff there.

```rust
use std::convert::Infallible;
use hyper::{Body, Request, Response, Server};
use hyper::service::{make_service_fn, service_fn};

async fn hello_world(_req: Request<Body>) -> Result<Response<Body>, Infallible> {
    Ok(Response::new("Hello, World".into()))
}

#[tokio::main]
pub async fn main() { 
    use std::net::SocketAddr;
    use std::thread;

    use futures::future;

    use tokio_modbus::prelude::*;
    use tokio_modbus::server::{self, Service};

    struct MbServer;

    impl Service for MbServer {
        type Request = Request;
        type Response = Response;
        type Error = std::io::Error;
        type Future = future::Ready<Result<Self::Response, Self::Error>>;

        fn call(&self, req: Self::Request) -> Self::Future {
            match req {
                Request::ReadCoils(_addr, cnt) => {
                    let mut coils = vec![false; cnt as usize];
                    coils[0] = true;
                    future::ready(Ok(Response::ReadCoils(coils)))
                },
                Request::ReadHoldingRegisters(_addr, cnt) => {
                    let mut registers = vec![0; cnt as usize];
                    registers[1] = 0x77;
                    future::ready(Ok(Response::ReadHoldingRegisters(registers)))
                }
 
                _ => unimplemented!(),
            }
        }
    }

    let mb_saddr  = SocketAddr::from(([0, 0, 0, 0], 502));
    let web_saddr = SocketAddr::from(([0, 0, 0, 0], 8080));

    println!("Starting up server...");

    let _server = thread::spawn(move || {
        server::tcp::Server::new(mb_saddr).serve(|| Ok(MbServer));
    });

        let web_service = make_service_fn(|_conn| async {
        Ok::<_, Infallible>(service_fn(hello_world))
    });

    let web_server = Server::bind(&web_saddr).serve(web_service);

    if let Err(e) = web_server.await {
        eprintln!("server error: {}", e);
    }
}
```

flosse, 2020-04-20:
@madprogrammer I added your code into the [`tcp-web-server` branch](https://github.com/slowtec/tokio-modbus/tree/tcp-web-server).
I'll hopefully adjust your example within this week but I can't promise.

madprogrammer, 2020-04-20:
@flosse Thank you! At the moment I temporarily gave up Rust implementation of my server and switched back to Python implementation, because I didn't figure out how to pass a Struct with shared global state to MbServer, so that I could use it in request handlers.

uklotzde, 2023-03-04:
I guess this issue is obsolete. Server tasks could be scheduled on any Tokio runtime by the application.

closed 2023-03-04 by uklotzde