slowtec/tokio-modbus #59: Socket not flushed before write - perpetual error once a read timeouts

MightyPork, 2020-05-15:
I'm using `tokio::time::timeout` on the Futures returned by e.g. `client.read_holding_registers()`, as real-world devices are not reliable and a retry is sometimes needed, also I don't want to block forever and freeze on one failing device.

When a device responds too late for a timeout, the response is queued in the TcpStream for reading.

Next time a read-write operation runs, it receives the OLD frame and errors out. Next time, it gets a newer old frame, ad infinitum.

```
Call ReadHoldingRegisters(0, 74)
Read error: Invalid response header: expected/request = Header { transaction_id: 67, unit_id: 12 }, actual/response = Header { transaction_id: 66, unit_id: 12 }
Retrying (retry 1/3)
Call ReadHoldingRegisters(0, 74)
Read error: Invalid response header: expected/request = Header { transaction_id: 68, unit_id: 12 }, actual/response = Header { transaction_id: 67, unit_id: 12 }
Retrying (retry 2/3)
Call ReadHoldingRegisters(0, 74)
Read error: Invalid response header: expected/request = Header { transaction_id: 69, unit_id: 12 }, actual/response = Header { transaction_id: 68, unit_id: 12 }
Retrying (retry 3/3)
Call ReadHoldingRegisters(0, 74)
Read error: Invalid response header: expected/request = Header { transaction_id: 70, unit_id: 12 }, actual/response = Header { transaction_id: 69, unit_id: 12 }
Retries exhausted!
```

I'm not sure where or how, but the socket needs to be flushed. I think a similar problem will be in the RTU client, but since it doesn't use transaction IDs, it might be harder to detect.

jlsky, 2022-07-01:
tokio::select! {
    result = ctx.read_holding_registers(3,12) => match result {
        Ok(v) => println!("Reading holding {:?}", v),
        Err(_) => println!("Reading holding err"),
    },
    _ = tokio::time::sleep(std::time::Duration::from_millis(100)) => println!("reading timeout 100 ms"),  
}

flosse, 2024-09-12:
I consider the issue to be resolved.
Feel free to open it again if it is not.

closed 2024-09-12 by flosse