slowtec/tokio-modbus #66: Panic when pulling cable on RTU device

Foelsgaard, 2020-10-21:
Pulling the cable on a device while a request is processing results in a panic at `tokio-modbus/blob/master/src/service/rtu.rs:60`:

```
    async fn call(&mut self, req: Request) -> Result<Response, Error> {
        let disconnect = req == Request::Disconnect;
        let req_adu = self.next_request_adu(req, disconnect);
        let req_hdr = req_adu.hdr;

        self.service.send(req_adu).await?;
        let res_adu = self.service.next().await.unwrap()?; // This is where it panics

        match res_adu.pdu {
            ResponsePdu(Ok(res)) => verify_response_header(req_hdr, res_adu.hdr).and(Ok(res)),
            ResponsePdu(Err(err)) => Err(Error::new(ErrorKind::Other, err)),
        }
    }

```

This seems to be a regression from v0.3.5 where it would instead return a broken pipe error.

It looks like it should be a pretty easy fix. Just map the `None` to some error and return that.

flosse, 2020-10-23:
@Foelsgaard thanks for reporting!
Hopefully I'll be able to fix that soon.

closed 2020-10-23 by uklotzde

Foelsgaard, 2020-10-23:
@flosse Great! Thank you so much! :)

PR #67 (2020-10-23): Fix: handle proken pipe in RTU service
should fix #66