slowtec/tokio-modbus #71: Upgrade to tokio 0.3

Baev1, 2020-12-15:
Hi!

Currently tokio is on version 0.3.6, while this crate only supports the 0.2 versions of tokio.

Using the 0.3 versions of tokio leads to a `thread 'tokio-runtime-worker' panicked at 'there is no reactor running, must be called from the context of Tokio runtime'` runtime error, when using the tcp client to connect to something.

flosse, 2020-12-21:
@Baev1 Thanks for reporting this. Unfortunately we currently have no capacity to update this crate.
So feel free to open a PR ;-)

Baev1, 2020-12-21:
@flosse thank you for your reply! 

I will look into making a PR to help with this, but after looking into things a bit more, I will wait for the soon upcoming 1.0 release of tokio (said to come at the end of this month) as announced back in October in their [blog](https://tokio.rs/blog/2020-10-tokio-0-3), as to prevent having to make another PR soon afterwards again.

flosse, 2020-12-21:
> I will look into making a PR to help with this,

great!
And don't hesitate to ask for help :)

uklotzde, 2020-12-21:
Afterwards we need to bump the version number to v0.5. Many crates still require tokio 0.2.x which might cause dependency conflicts in a project? Not sure about this.

But because of these potential conflicts we should prepare ourselves for the upgrade. You could start with tokio 0.3.x to determine the current gap. 1.0 was announced to released before the end of this year.

Baev1, 2020-12-21:
I skimmed through the dependencies and it seems to me `tokio-serial` will be a blocker for this, as it also uses tokio 0.2. In addition tokio-serial uses mio-serial, which depends on mio 0.6, while afaik tokio 0.3 uses mio 0.7 and I am not sure if that would then in turn block tokio-serial from using tokio 0.3 or not. 

I can't with confidence say if this all means we can or can't upgrade tokio here, but yeah it is be likely that I first have to take a look at making a PR for tokio-serial to bump their tokio version.

Also, unrelated, but I noticed that `net2` is deprecated in favour of `socket2`.

llamicron, 2021-02-24:
Has there been any progress on this? More popular crates have been updating to `tokio 1.0` or at least `tokio 0.3`, which makes integration with this difficult. Is this still on the roadmap?

flosse, 2021-03-04:
@llamicron I'm quite busy right now but feel free to continue with #72 :smiley:

PR #78 (2021-07-28): Yet another Tokio v1 PR
Based on the PR from @Tarnadas but with `tokio_serial` added back in as a dependency instead of `serial-io` due to its incoming Windows pipes support (from https://github.com/berkowski/tokio-serial/pull/42 and earlier work).
Also homogenized and expanded a bit of documentation for a little bit.

Works on my machine™ (actually semi-field-tested against a serial device from a Linux host and communications appear to work).
Reviews and eventual merge are welcome.

Closes #71 and #76
PR #76 (2021-07-28): tokio v1
Hey,

here is another attempt at updating tokio to v1.
Some remarks:

- I needed to enable the "rt-multi-thread" feature. Otherwise the tcp server would just block and nothing happens.
- I replaced `tokio-serial` with `serial-io`, a custom fork I made based on `mio-serial` and `tokio-serial`. It basically just adds tokio v1 support via a feature flag instead of splitting it into different crates. Also all other dependencies have been updated. I checked [this PR](https://github.com/berkowski/tokio-serial/pull/33) and made sure that I [fixed this](https://github.com/berkowski/tokio-seri [... cut]
closed 2021-07-28 by flosse