slowtec/tokio-modbus #77: tcp-server.rs tag v0.4.0 doesn't compile

jpinafcirce, 2021-06-14:
Hello,

Last week (11th June) I have been testing 'tcp-server.rs' from 'slowtec/tokio-modbus/examples' and there is some kind of issue related to 'tcp' and 'server' features. If you try to compile 'tcp-server.rs' you will receive an error and I think that it comes from the feature 'server'. It is like it cannot be found even writing it in 'dependencies' in 'Cargo.toml'

However, I tried 'tcp-server.rs' tag v0.3.5 and it runs without problem.

bootrecords, 2021-07-23:
The `server::tcp` module is gated by the `tcp-server-unstable` feature, as the CHANGELOG for versio 0.4.0 indicates.

Accordingly, you'll get the following error on trying `cargo run --features=tcp,server --example tcp-server`:

```
error[E0433]: failed to resolve: could not find `tcp` in `server`
  --> examples/tcp-server.rs:34:17
   |
34 |         server::tcp::Server::new(socket_addr).serve(|| Ok(MbServer));
   |                 ^^^ could not find `tcp` in `server`

error: aborting due to previous error
```

The example will run fine with `cargo run --features=tcp-server-unstable --example tcp-server`, however.

TBH, I think this extra feature is pretty much redundant, as the `server` module is already gated by the `server` feature (which is also activated by the `tcp-server-unstable` feature), while the `server::tcp` module maybe should better be guarded by simply the `tcp` feature instead.

But I guess @flosse may have had in mind to make the server functionality opt-in even more explicit via that feature flag.

flosse, 2021-07-28:
> But I guess @flosse may have had in mind to make the server functionality opt-in even more explicit via that feature flag.

Exactly!

We haven't really used/tested the server implementation in either a production or demo context, so you really shouldn't use it unless you're playing around with it for fun.

But you are welcome to open a PR with a production-ready implementation ;-)

closed 2021-07-28 by flosse

bootrecords, 2021-07-28:
@flosse we've used the server-side implementation at least in a staging environment for a while at Volterion - though that was still based on `tokio-modbus` 0.3 - I don't really know if they updated the version in the meantime, but at least functionality-wise other applications like [modpoll](https://www.modbusdriver.com/modpoll.html) and [Modbus Poll](https://www.modbustools.com/modbus_poll.html) had no trouble accessing Modbus registers. Same goes for at least the official SunSpec client software based on [pysunspec](https://github.com/sunspec/pysunspec).

flosse, 2021-07-28:
@bootrecords thanks for sharing this!
Do I understand you correctly that you'd mark the server implementation as production ready?

bootrecords, 2021-07-29:
@flosse I'm not sure about your production-readiness criteria and what edge cases and issues you'd expect to be tested for that, but from my experience the Modbus server implementation there was at least usable and running reliably based on `tokio-modbus 0.3` - you may want to reach out to Volterion and ask them about their experience with the library to date, though from what little I know about their application at this point (given that I've left that company almost a year ago) they still haven't upgraded to version `0.4`.

uklotzde, 2021-07-29:
The server implementation should only be provided as an example. Then it could be used as a blueprint for your custom server.

We could provide the building blocks, but a method like `server::tcp::serve_until()` doesn't belong into the public API. Too many batteries included. It already requires to enable unneeded dependencies of tokio, see the comment in Cargo.toml.

uklotzde, 2021-07-29:
#81 might resolve the dependency/feature issues with the server part.

flosse, 2021-07-29:
> The server implementation should only be provided as an example.

So then we should move the code to the example folder, right?