slowtec/tokio-modbus #80: Release v0.5.0

flosse, 2021-07-28:
- [ ] Wait for [`tokio-serial`](https://github.com/berkowski/tokio-serial) `v5.4.0` release
- [ ] Update [`CHANGELOG.md`](https://github.com/slowtec/tokio-modbus/blob/master/CHANGELOG.md)
- [ ] Publish on [`crates.io`](https://crates.io)

uklotzde, 2021-07-29:
How about removing `sync` from the default features?

uklotzde, 2021-08-13:
> How about removing `sync` from the default features?

#89

flosse, 2021-08-20:
done :)

closed 2021-08-20 by flosse

bootrecords, 2021-08-20:
@flosse great work, though I'd say you forgot to mention the addition of an RTU server implementation in the Changelog.

I also noticed that the traits that use `async_trait` now are also exported and can be implemented by library users, which then also have to declare `async_trait` in their usage. That shouldn't be much of an issue, since `async_trait` is already in the depencies anyway, but I guess it would help to add some documentation somewhere to explain proper usage of those traits if anyone intends to make use of them (I may set up an according PR over the weekend, given that I mainly transitioned those traits to `async_trait` in the first place)