slowtec/tokio-modbus #98: Any example for ExceptionResponse?

ArcherGu, 2021-08-30:
I'm a rust newcomer and am building a modbus server, and when the client sends the wrong address I need to give feedback on the error message, whether there are any examples of ExpressinResponse

justinyhuang, 2021-11-29:
Any response to this question? it appears to me that the exception definitions are in `frame` which is private.
How could a server implementation produce an appropriate exception response in this case?

glennpierce, 2022-01-17:
I wanted to match on 

Custom { kind: Other, error: ExceptionResponse { function: 3, exception: GatewayTargetDevice } } => {

}

but you are right frame is private where I find myself needing a few things from it.
In the end I converted the error to a string and did a match (yuck)

LenRemmerswaal, 2022-03-18:
I would think that you could have the modbus server return an error response, by returning  from call() like so:
```
 . . . 
    let fncode = req_to_fn_code(&req);
    return std::future::ready(Err(ExceptionResponse {
        function: fncode,
        exception: Exception::IllegalFunction,   // Or any other kind of Modbus exception
    }));
 . . .
```
Like mentioned before, for this to work, the following needs to be published:
- frame::ExceptionResponse
- frame::Exception
- codec::req_to_fn_code()

The first 2 could be included in the prelude, like Request and Response are.
The last 1 I am not sure about.

If you return with another kind of error than an ExceptionResponse, like std::io::Error, the application compiles, but the connection is forcibly closed by the server when the error is returned.

closed 2024-09-12 by uklotzde