slowtec/tokio-modbus #100: Sync Client: could not find `runtime` in `tokio`

Hywan, 2021-11-15:
Hello,

Thank you for this nice project :-).

Trying the small project with tokio-modbus 0.5.0 raises a compilation error:

```toml
[dependencies]
tokio-modbus = { version = "0.5", default-features = false, features = ["tcp", "sync"] }
```

```rs
pub fn main() {
    use tokio_modbus::prelude::*;

    let socket_addr = "192.168.0.222:502".parse().unwrap();
    let mut ctx = client::sync::tcp::connect(socket_addr).unwrap();
    let buff = ctx.read_input_registers(0x1000, 7).unwrap();
    println!("Response is '{:?}'", buff);
}
```

It raises the following error:

```sh
cargo run
   Compiling tokio-modbus v0.5.0
error[E0433]: failed to resolve: could not find `runtime` in `tokio`
  --> /Users/hwhost/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-modbus-0.5.0/src/client/sync/mod.rs:50:18
   |
50 |     core: tokio::runtime::Runtime,
   |                  ^^^^^^^ could not find `runtime` in `tokio`

error[E0433]: failed to resolve: could not find `runtime` in `tokio`
  --> /Users/hwhost/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-modbus-0.5.0/src/client/sync/tcp.rs:16:30
   |
16 |     let rt = tokio::runtime::Builder::new_current_thread()
   |                              ^^^^^^^ not found in `tokio::runtime`
   |
help: consider importing one of these items
   |
1  | use std::thread::Builder;
   |
1  | use tokio_util::codec::length_delimited::Builder;
   |

For more information about this error, try `rustc --explain E0433`.
error: could not compile `tokio-modbus` due to 2 previous errors
```

Can you reproduce this error? It seems to happen only with the `sync` feature.

flosse, 2021-11-16:
> Can you reproduce this error?

yes!

Thanks for reporting!

#101 Hopefully fixes this.

flosse, 2021-11-16:
@Hywan #101 is blocked because of security issues with `serialport-rs` that is unmaintained at the moment.
In the meantime you can do this in your `Cargo.toml`:

```toml
[dependencies.tokio-modbus]
git = "https://github.com/slowtec/tokio-modbus"
rev = "0421a8b99a85c80fe926ad3a0dfacc14b5b8f8a2"
default-features = false
features = ["tcp", "sync"]
```

flosse, 2021-11-21:
@Hywan I just merged the fix into the master, so the next published version should work for you as expected. If that is not the case, please reopen this issue :)

closed 2021-11-21 by flosse

Hywan, 2021-12-01:
0.5.1 fixes the problem, thanks!

PR #101 (2021-11-21): Fix: require tokio/rt for sync feature
Should fix #100