slowtec/tokio-modbus #103: How to handle unmaintained serial dependencies?

flosse, 2021-11-18:
The issue started here:
[serialport-rs](https://gitlab.com/susurrus/serialport-rs/) uses an outdated (`v0.16`) `nix`crate :disappointed: 

And this is the current (2021-11-18) situation:
- serialport-rs is [currently not maintained](https://gitlab.com/susurrus/serialport-rs/-/issues/111).
-  [mio-serial](https://github.com/berkowski/mio-serial) is not maintained anymore
-  [tokio-serial is currently not maintained](https://github.com/berkowski/tokio-serial#notice)
 
This is a first move
- There is a [PR](https://gitlab.com/susurrus/serialport-rs/-/merge_requests/107) to update `nix` at least to `v0.18`
- I made a [PR](https://github.com/Susurrus/serialport-rs/pull/3) to update `nix` to `v0.23.x`
- The [`updates` branch](https://github.com/slowtec/serialport-rs/tree/updates) @slowtec of `serialport-rs` has some more updates

ssnover, 2021-11-18:
If Slowtec isn't looking to take on indefinite maintenance of serialport-rs, I'd note that it looks like it's maintenance is being carried on here: https://gitlab.com/reconbot/serialport-rs (based on the thread linked). It may be worth making an additional PR to that repo and see if there's a response.

lindblandro, 2023-03-25:
The thread now mentions https://github.com/serialport/serialport-rs and that seems fairly active as the most recent commits are from just a few days ago.