slowtec/tokio-modbus #107: Cannot run RTU client examples

sirhcel, 2021-12-15:
* I wanted to give to give the Modbus RTU client examples a test drive but failed in both cases
    * [`rtu-client-sync.rs`](https://github.com/slowtec/tokio-modbus/blob/5bd83cf660cf79edfe579f91a168e924e73553aa/examples/rtu-client-sync.rs) panics with
        ```
        $ cargo run --features rtu,sync --example rtu-client-sync
           Compiling tokio-modbus v0.5.2 (/.../tokio-modbus)
            Finished dev [unoptimized + debuginfo] target(s) in 0.85s
             Running `target/debug/examples/rtu-client-sync`
        thread 'main' panicked at 'there is no reactor running, must be called from the context of a Tokio 1.x runtime', /.../.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.14.0/src/runtime/context.rs:29:26
        note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
        ```
    * [`rtu-client.rs`](https://github.com/slowtec/tokio-modbus/blob/5bd83cf660cf79edfe579f91a168e924e73553aa/examples/rtu-client.rs) hangs attempting to read data
        ```
        $ cargo run --features rtu --example rtu-client
            Finished dev [unoptimized + debuginfo] target(s) in 0.06s
             Running `target/debug/examples/rtu-client`
        Reading a sensor value
        ```
* Are these examples expected to run out of the box?
* What could I do to actually get them running?
* At a first first glance into [`context.rs:29`](https://github.com/tokio-rs/tokio/blob/f64673580dfc649954eb744eb2734f2f118baa47/tokio/src/runtime/context.rs#L29) it looks like borrowing from `CONTEXT` fails and I bet there is some initialization missing. But this is my first contact with Tokio and I have no idea about the what, how, and why.

sirhcel, 2021-12-16:
* I meanwhile managed to get [`rtu-client-sync.rs`](https://github.com/slowtec/tokio-modbus/blob/5bd83cf660cf79edfe579f91a168e924e73553aa/examples/rtu-client-sync.rs) running explicitly bringing in a Tokio runtime:
    ```rust
    use tokio::runtime::Runtime;
    fn main() -> Result<(), Box<dyn std::error::Error>> {
        let rt = Runtime::new().unwrap();
        let _guard = rt.enter();

        println!("Hello, world!");
        // ...
    }
    ```
* What's the right way, to get the sync example running?
* I initially erroneously reported the async example hanging (I used the wrong parity configuration) and updated title and description accordingly

manio, 2021-12-22:
Maybe the example is outdated?
What if you add the line similar as in this example:
https://github.com/slowtec/tokio-modbus/blob/5bd83cf660cf79edfe579f91a168e924e73553aa/examples/tcp-client-custom-fn.rs#L1
in the first line before main, so it will start the tokio runtime.
Is it OK then?

sirhcel, 2021-12-22:
I already tried this, but this still results in the same error message as shown in the description. My example looks like:
```diff
$ git diff examples/rtu-client-sync.rs
diff --git a/examples/rtu-client-sync.rs b/examples/rtu-client-sync.rs
index 5872f26..cf11c96 100644
--- a/examples/rtu-client-sync.rs
+++ b/examples/rtu-client-sync.rs
@@ -1,4 +1,5 @@
-pub fn main() -> Result<(), Box<dyn std::error::Error>> {
+#[tokio::main(flavor = "current_thread")]
+pub async fn main() -> Result<(), Box<dyn std::error::Error>> {
     use tokio_modbus::prelude::*;

     let tty_path = "/dev/ttyUSB0";
```
I would have expected this to work as well.

sirhcel, 2021-12-28:
What is the "official" way for providing a runtime for this example? If it has to be done explicitly like in https://github.com/slowtec/tokio-modbus/issues/107#issuecomment-996261254, I will create a PR for updating it.

vita1it4, 2022-06-16:
Did someone manage to run an example of reading modbus in synchronous mode? It doesn't work for me either, same error as described in the first post. The asynchronous read example works, but I need a synchronous one.

lanquil, 2022-06-17:
I did manage as suggested in https://github.com/slowtec/tokio-modbus/issues/107#issuecomment-996261254

vita1it4, 2022-06-21:
This example doesn't work for me:
```
   |
10 |     let rt = Runtime::new().unwrap();
   |                       ^^^ function or associated item not found in `Runtime`
```

lanquil, 2022-06-21:
> This example doesn't work for me:
> 
> ```
>    |
> 10 |     let rt = Runtime::new().unwrap();
>    |                       ^^^ function or associated item not found in `Runtime`
> ```

Have you brought into scope the correct Runtime object? `use tokio::runtime::Runtime;`

Can you post your code?

vita1it4, 2022-06-21:
Here is my code:
```
//use tokio_serial::{DataBits, Parity, StopBits};
//use tokio_modbus::prelude::*;
use tokio::runtime::Runtime;

fn main() -> Result<(), Box<dyn std::error::Error>> {
    let rt = Runtime::new().unwrap();
    let _guard = rt.enter();

    println!("Hello, world!");
/*
    let slave = Slave(4);
    let builder = tokio_serial::new("/dev/ttyUSB0", 19200);
    let builder = builder
        .data_bits(DataBits::Eight)
        .stop_bits(StopBits::Two)
        .parity(Parity::None);
*/
//    let mut ctx = sync::rtu::connect_slave(&builder, slave)?;
//    println!("Reading a sensor value");
//    let rsp = ctx.read_input_registers(25, 1)?;
//    println!("Sensor value is: {:?}", rsp);
    Ok(())
}
```

lanquil, 2022-06-21:
I think you are missing tokio `rt-multi-thread` feature.

I have this in Cargo.toml:
```toml
tokio-modbus = { version = "0.5.2", default-features = false, features = [
    "rtu",
    "sync",
] }
tokio-serial = "5.4.2"
tokio = { version = "1.17.0", features = ["rt-multi-thread"] }
```

vita1it4, 2022-06-21:
Yes you are right. I'm missing tokio `rt-multi-thread` feature. 
Now that I've added it, my code is working. And synchronous reading modbus now also works.
Thank you!

sirhcel, 2022-06-22:
#116 resolves this and the "vanilla" example works now out of the box.

closed 2022-06-22 by sirhcel