slowtec/tokio-modbus #112: Maximum number of input registers read

chengchangwu, 2022-04-11:
I've found that I can read at most 126 input registers with `read_input_registers()`. But why 126? Which part of the source code should I read.

chengchangwu, 2022-04-12:
But I can read 126 contiguous registers. one more thsn the spec.

uklotzde, 2023-03-04:
This bug affects multiple function codes. Currently there are no semantic, client-side parameters according to the Modbus protocol validations in place. If a parameter value could technically be serialized it is accepted, even if the Modbus protocol specifies more restrictive, semantic bounds. Ideally servers should reject such non-standard requests.

Not a severe bug as clients are responsible for sending valid requests and servers are advised to reject invalid requests. Enforcing strict standard behavior by the library may even prevent using custom, vendor-specific extensions.

uklotzde, 2023-03-05:
Not a bug, it's a "feature".

If those additional conformance checks are implemented then they should be hidden behind a feature flag. Not everyone might need or want them.

uklotzde, 2023-03-06:
This affects both the client and the server.