slowtec/tokio-modbus #114: Attempting basic client sync example failed

saturn77, 2022-04-29:
In attempting to use the client-sync example, I simply cannot get it to work and was wondering if there is some library that need to include . . . Thanks, James

[image]

uklotzde, 2022-04-29:
Please get familiar with the Rust module system: https://doc.rust-lang.org/book/ch07-04-bringing-paths-into-scope-with-the-use-keyword.html

saturn77, 2022-04-29:
The example will not compile, even with the Cargo.toml file set to include tokio-modbus. I have been using Rust for many years. This module is broken, and your repository does not have any continuous integration testing.

uklotzde, 2022-04-30:
Ups, sorry. My fault, didn't look closely at the code and your previous experience :see_no_evil: 

The code you posted is in fact copied from one of the examples. All examples compile and work, verified by 2 CI workflows, namely [pre-commit](https://github.com/slowtec/tokio-modbus/blob/master/.github/workflows/pre-commit.yaml) and [continuous_integration](https://github.com/slowtec/tokio-modbus/blob/master/.github/workflows/continuous_integration.yaml). I checked both v0.5.2 and the main branch.

Did you enable both the `sync` and `tcp` features?

uklotzde, 2022-04-30:
I just notices that the continuous_integration workflow is missing the `--all-targets` target selection. The examples are already checked in the pre-commit workflow by clippy, so this is not expected to reveal any hidden errors.

saturn77, 2022-05-01:
Thanks for the feedback uklotzde. 

I was able to get the tokio-modbus library to work for me in a design and there were a couple of issues. 

1) Morphed the example code for tcp-sync with the  code in the document for tcp-sync and employed ? function
2) Added "features" of  sync and tcp in the Cargo.toml file ( Didn't see this in the Cargo.toml previoulsy)
3) Adjusted to using connect_slave as shown below. 

After this, I have been able to connect successfully. 

Thanks. James

[image]

closed 2022-05-03 by saturn77