slowtec/tokio-modbus #126: Performance compared to libmodbus?

ShayBox, 2022-10-07:
Has there been any benchmarks done compared to `libmodbus` or any other high performance modbus libraries for Rust?

CosminPerRam, 2022-10-08:
Thanks for doing this benchmark, happy to hear that `tokio_modbus` has absolute minimal (or non existent? every result is +/- 0.5% max (which I think is normal)) overhead against `libmodbus`.

ShayBox, 2022-10-09:
It does appear to have no difference, however I found I was able to achieve a step higher baud rate with Tokio mod us over libmodbus so it is faster for me. But I'm going over my serial devices max baud rate and encountering expected issues.

uklotzde, 2023-03-04:
Closing as completed as there is nothing left to do here.

If you think that benchmarks should be added to the project then please submit a PR.

closed 2023-03-04 by uklotzde