slowtec/tokio-modbus #128: More documentation for the server features

raoulspronck, 2022-10-25:
Hi there, 

I want to build an TCP modbus server to be able to connect multiple robots and machines in our machine park. I came across this tokio-modbus library for rust. I followed the examples to build an TCP Server but I'm missing some documentation about all the functions and how to implement them. 

 got this from the documentation but is it really reading an register or do I need to implement this on my own?

```
impl Service for MbServer {
    type Request = Request;
    type Response = Response;
    type Error = std::io::Error;
    type Future = future::Ready<Result<Self::Response, Self::Error>>;
    fn call(&self, req: Self::Request) -> Self::Future {
        println!("{:?}", req);
        match req {
            Request::ReadInputRegisters(_addr, cnt) => {
                let mut registers = vec![0; cnt.into()];
                registers[2] = 77;
                future::ready(Ok(Response::ReadInputRegisters(registers)))
            }
            _ => unimplemented!(),
        }
    }
}
```

How do I write an input register ? Do I do it like this or? 

```
Request::WriteSingleRegister(addr, cnt) => {
    let adress = addr.into();
    let register = cnt.into();
    future::ready(Ok(Response::WriteSingleRegister(adress, register)))
}
```
And how do you implement all the other functions like reading/writing coils, ....

I like how easy this library was to set up compared to others but I'm missing some documentation or some more extensive examples. 

Maybe it is just me and I'm sorry if this all sounded dumb but can anyone help me?

Thanks in advance!

cdbennett, 2023-02-24:
Agreed, would be great to have some better documentation and examples.

I am looking at using this library as both a client and as a server.  I'd be willing to help contribute to some examples.

Sounds like the server code is going to be extracted into a separate crate.  See #146 for the tracking issue. (It's behind a `tcp-server-unstable` feature flag already, which is a sign that something is up.)

uklotzde, 2023-02-25:
Welcome @cdbennett.

Since we have no use case with actual requirements and are also short on resources (time and money) there is nothing "up". #146 only documents my and our opinion on how to proceed in the future.

If you are able to prove that the server traits are suitable for more than just implementing some toy examples and could provide feedback for improving both implementation and documentation that would be great.

cdbennett, 2023-03-07:
I started expanding the tcp-server example in #163 .  I'm trying to figure out how to send and receive Modbus exception responses with specific code handling, e.g. invalid function code, invalid data address.  Maybe the means to do this is not exposed publicly in the API yet.  Or maybe I just haven't figured it out yet.