slowtec/tokio-modbus #133: MODBUS/TCP Security support

gustavowd, 2022-11-10:
Hi. Congrats for the project.
Is it possible to add support for the new specification, the Modbus TCP Security, based on TLS?

Some of new protocols based on modbus are migrating for this modbus version, as the sunspec modbus protocol. The specification is available at:
[https://modbus.org/docs/MB-TCP-Security-v21_2018-07-24.pdf]

I beleive it can be implemented based on tokio-rustls.

Thanks in advance.
Gustavo Denardin

gustavowd, 2023-02-23:
Hi [uklotzde](https://github.com/uklotzde),

I'm new to the rust development, but I succeeded in implement a simple tls server using some crates from the tokio modbus project. However i do not complete understand how to integrate this tls server to the project code in order to do a pull request. I will post the code here to contribute with the project.

I have tested the server with a client written with C and libmodbus. It's working.

main.rs:
use argh::FromArgs;
use rustls_pemfile::{certs, ec_private_keys};
use std::fs::File;
use std::io::{self, BufReader};
use std::net::ToSocketAddrs;
use std::path::{Path, PathBuf};
use std::sync::Arc;
use std::convert::From;
use tokio::net::TcpListener;
use tokio_rustls::rustls::{self, Certificate, PrivateKey};
use tokio_rustls::TlsAcceptor;
use tokio_util::codec::{Framed};
use futures_util::{future, sink::SinkExt as _, stream::StreamExt as _};

mod slave;
mod codec;
mod frame;
mod service;

use crate::frame::{Response, ResponsePdu, ExceptionResponse, Request};

/// Tokio Rustls server example
#[derive(FromArgs)]
struct Options {
    /// bind addr
    #[argh(positional)]
    addr: String,

    /// cert file
    #[argh(option, short = 'c')]
    cert: PathBuf,

    /// key file
    #[argh(option, short = 'k')]
    key: PathBuf,
}

fn load_certs(path: &Path) -> io::Result<Vec<Certificate>> {
    certs(&mut BufReader::new(File::open(path)?))
        .map_err(|_| io::Error::new(io::ErrorKind::InvalidInput, "invalid cert"))
        .map(|mut certs| certs.drain(..).map(Certificate).collect())
}

fn load_keys(path: &Path) -> io::Result<Vec<PrivateKey>> {
    ec_private_keys(&mut BufReader::new(File::open(path)?))
        .map_err(|_| io::Error::new(io::ErrorKind::InvalidInput, "invalid key"))
        .map(|mut keys| keys.drain(..).map(PrivateKey).collect())
}

#[tokio::main]
async fn main() -> io::Result<()> {
    let options: Options = argh::from_env();

    let addr = options
        .addr
        .to_socket_addrs()?
        .next()
        .ok_or_else(|| io::Error::from(io::ErrorKind::AddrNotAvailable))?;
    
    let certs = load_certs(&options.cert)?;
    let mut keys = load_keys(&options.key)?;
    //println!("{:?}", keys);
    let config = rustls::ServerConfig::builder()
[... 73 lines cut ...]
            }
        });
    }
}

cargo.toml:
[package]
name = "tokio-tls-server"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
futures = "0.3.26"
futures-util = "0.3.26"
tokio = { version = "1.25.0", features = [ "full" ] }
argh = "0.1.10"
rustls-pemfile = "1.0.2"
tokio-rustls = "0.23.4"
byteorder = "1.4.3"
bytes = "1.4.0"
tokio-util = { version = "0.7.7", default-features = false, features = [
    "codec",
] }

gustavowd, 2023-02-23:
As I understand it, this code can help with the development of a TLS server for the project, as it demonstrates how to create a TLS connection, accept new connections, and receive and send packets over TLS.

uklotzde, 2023-03-06:
The client-side support for TLS still needs to be resolved. I missed that aspect.

To keep thinks manageable I created a new, focused issue #161 and added a back link to this issue.

uklotzde, 2023-03-06:
I noticed that we still need to abstract over both `TlsStream` and `TcpStream` to achieve an agnostic implementation for both client and server. My bad.

uklotzde, 2023-03-14:
The addition of both TLS client and server examples demonstrate that TLS support could be added as an additional layer on top of the foundational Modbus code.

A fully configurable, _batteries included_ approach enabled by a Cargo feature would be handy. But as long as no one is able to provide a solid and flexible implementation that is suitable for covering (almost) all practical use cases we should close this issue.

PR #159 (2023-03-06): Revise server features
The current design is both too opinionated and inflexible.

- Stabilizes "tcp-server-unstable" as "tcp-server"
- Obsoletes #134
- Fixes #133
- Obsoletes #151
- Fixes #115
closed 2023-03-14 by uklotzde