slowtec/tokio-modbus #144: Wrapper around tokio-modbus sync/async

ShayBox, 2023-02-03:
I'm making a wrapper around tokio-modbus for a device to make it easier to interact with it, but I want to support both sync and async, what's the easiest way to do that?

I was thinking maybe-async but since tokio-modbus supports both, maybe there's a better way.

ShayBox, 2023-02-04:
Also, you're using the global `sync` namespace `sync::tokio_modbus`, it should be `tokio_modbus::sync`.

uklotzde, 2023-02-05:
> Also, you're using the global `sync` namespace `sync::X`, it should be `tokio_modbus::sync::X`.

I don't understand this comment.

uklotzde, 2023-02-05:
`tokio-modbus`: The _sync_ API is just an opinionated wrapper around the _async_ API for convenience.

Better only use the _async_ API if you want to provide an API on a higher level of abstraction and provide your own custom _sync_ wrapper around your API.

closed 2023-02-05 by uklotzde

ShayBox, 2023-02-05:
> > Also, you're using the global `sync` namespace `sync::X`, it should be `tokio_modbus::sync::X`.
> 
> I don't understand this comment.

Never mind, it was the prelude bringing `sync` into scope not a global namespace from the package.

> `tokio-modbus`: The _sync_ API is just an opinionated wrapper around the _async_ API for convenience.
> 
> Better only use the _async_ API if you want to provide an API on a higher level of abstraction and provide your own custom _sync_ wrapper around your API.

I plan to support both async and sync, but I would like to pass the sync onto your library as you support sync instead of blocking the async library myself.

Looking at the sync I see it pretty much does the same thing that `maybe-async` would do, you could remove the sync code and use it to generate sync from async.