slowtec/tokio-modbus #155: crash when using sync::rtu with timeout

vfiack, 2023-02-28:
This code doesn't set a timeout, even if SerialPortBuilder timeout is set:

```
    let builder = tokio_serial::new(config.serial_port(), config.baud_rate())
        .timeout(Duration::from_secs(3))
        .parity(Parity::None)
        .data_bits(DataBits::Eight)
        .stop_bits(stop_bits);
    let mut context = sync::rtu::connect(&builder)?;
```

We can however set the timeout manually in the context:
```
    context.set_timeout(Duration::from_secs(3));
```

The problem is that it crashes later when calling block_on_with_timeout with:
```panicked at 'A Tokio 1.x context was found, but timers are disabled. Call `enable_time` on the runtime builder to enable timers.'```

The most probable fix would be to add enable_time when creating the runtime in client/sync/rtu.rs, connect_slave_with_timeout:
```
    let runtime = tokio::runtime::Builder::new_current_thread()
        .enable_io()
        .enable_time() // add this here
        .build()?;
```

uklotzde, 2023-02-28:
Thanks for testing.

I agree, unconditionally enabling timers in the implicitly created runtime is needed then.

uklotzde, 2023-02-28:
I'll prepare a bugfix release.

closed 2023-02-28 by uklotzde

vfiack, 2023-02-28:
Thanks, that was fast!

PR #156 (2023-02-28): sync: Fix panic when using client-side timeouts
- v0.7.1
- Fixes #155 
- tcp-server-unstable: Upgrade socket2 dependency (marked as _unstable_, so this possibly breaking change doesn't matter)