slowtec/tokio-modbus #169: Client: how to identify Modbus exception code sent by server in response

cdbennett, 2023-03-10:
How can user code using tokio-modbus as a Modbus client detect what specific exception code a server returned?

Two interrelated parts of this question:

1. How user code can distinguish between a recoverable error during a Modbus request, (e.g. IllegalDataAddress, and a connection failure, e.g. TCP connection closed by remote host, or a protocol framing error.)
2. How can user code identify which specific Modbus exception code was returned by the server. (E.g. IllegalDataAddress vs IllegalFunction)

In the case of Modbus exception responses, we don't want to close the Modbus connection. But if the network connection was lost, then user code needs to re-create a new tokio-modbus connection and start over.

Related: 
- https://github.com/slowtec/tokio-modbus/pull/163
- https://github.com/slowtec/tokio-modbus/issues/165

Stupremee, 2023-08-25:
Hey :wave: 

I'm really in the need of finding out what Modbus error I got from a slave for my application.
I noticed it's not possible using `tokio-modbus` yet (at leat without doing silly `err.to_string().contains("illegal data address")`, and co), and found this issue.

After looking through the code. I found a fast, simple way to get access to the exception from a modbus server. You just have to make the `ExceptionResponse` and `Exception` types publicy accessible. Then someone using this crate can try to downcast the `std::io::Error` to the `ExceptionResponse` type to extract the error.

cdbennett, 2024-08-21:
This issue is resolved, verified working now in tokio-modbus v0.14.0.

Example of issuing a command like:

```rust
    let values = read_registers(connection, start, count)
    .await
    .context("communication error")?
    .context("received a Modbus exception response")?;
```

and then the program might produce either a Modbus exception response:

```text
Error: received a Modbus exception response

Caused by:
    Gateway target device failed to respond
```

or some other I/O error (in this case simulated by connecting to a server that sends random bytes instead of Modbus frames back):

```text
Error: communication error

Caused by:
    Connection reset by peer (os error 104)
```

Thanks for getting this working!

closed 2024-08-21 by cdbennett