slowtec/tokio-modbus #177: rtu over tcp

pnpdb, 2023-03-23:
Is there any implementation of modbus rtu over tcp?

pnpdb, 2023-03-24:
[image]

Here is a simple explanation about RTU over TCP

pnpdb, 2023-03-24:
http://www.simplymodbus.ca/TCP.htm
Here is a simple explanation about RTU over TCP.
[image]

imawaki, 2023-04-17:
@pnpdb does this question relate? https://github.com/hirschenberger/modbus-rs/issues/37
@uklotzde is it difficult to implement?

uklotzde, 2023-04-17:
Before thinking about possible implementations I recommend to start evaluating the implications for the API and how it needs to be changed or extended. Then open a Draft PR for discussion.

cdbennett, 2023-07-27:
Could be interesting to be able to do this. But, seems like quite an edge case, how many people actually would use this? Is this the best way to really solve the problem you have? Maybe you could explain the bigger picture.

The way I've usually seen access to Modbus RTU devices done recently is to use a Modbus TCP to Modbus RTU gateway device. Typically this is a little embedded hardware device costing $100-$1000. But it could be a pure software component too that has a Modbus TCP server and Modbus RTU client in one application -- this would be more general purpose and clean, I think. We could implement such a software gateway using tokio-modbus.

wheregmis, 2023-08-16:
I am currently trying to combine server and client into one application, the purpose is different in my case but is doable. I am currently bridging multiple tcp sources under one server. If my workspace will have anything similar kind of use case, i will try to put some more insights.

barisdigi, 2023-08-18:
This is the issue preventing us from using tokio-modbus, some gateway devices unfortunately only support modbus rtu over tcp

uklotzde, 2023-08-18:
> This is the issue preventing us from using tokio-modbus, some gateway devices unfortunately only support modbus rtu over tcp

Then why don't you implement the missing pieces and and submit a PR?

HrvojeJuric, 2023-10-25:
u can connect rtu over tcp like this

```rust
let transport = TcpStream::connect(id.socket).await?;
tokio_modbus::prelude::rtu::attach_slave(transport, slave)
```

this works for me

uklotzde, 2024-01-03:
Does the server-side implementation added in #228 fix this issue or is the client-side part missing?

uklotzde, 2024-01-03:
I will close this issue. Please create a new, more specific issue for any missing parts.

closed 2024-01-03 by uklotzde