slowtec/tokio-modbus #179: How to implement a bus monitor

hellow554, 2023-04-01:
I have a modbus device that I'm trying to understand, because there's no public documentation available.

My current approach is to have a `Server` and dump all requests that it is receiving and returning `future::ready(Ok(None))` to not send an answer.

First question: Is that the right approach to not send any data back?

Second question: How do I receive responses on the bus?

Thanks for your help!

cdbennett, 2023-07-27:
So let me see if I understand correctly, you have a Modbus client device which is undocumented, and you want to see what kind of Modbus requests that client device makes to a server?

hellow554, 2023-07-28:
Yes, exactly.
I want to monitor both, the communication from the RTU to the device in question and back.

cdbennett, 2023-09-05:
You could implement a Modbus server using tokio-modbus, and in response to any Read/Write register commands you could just return zero bytes or whatever data you want to make up.  If you want the client device to send you other requests you probably don't want to "not send any data back".

Actually I want to clarify. At first It thought you only had the client device and you didn't have a corresponding server device it was talking to.

But, are you saying you already have both a "client" (master) and a "server" (slave) device, that are communicating correctly using some unknown protocol?  And you want to monitor their communications and inspect the messages?  Because that would suggest a different solution than if there were only one side of the pipe already.