slowtec/tokio-modbus #184: Infinite retry loop with heavy CPU load

ColinFinck, 2023-04-17:
I have tested tokio-modbus on a misbehaving Modbus RTU bus, which currently responds with an infinite stream of zeros to all read requests. This is not a theoretical situation, but can happen when a single device on the bus is misconfigured.
I don't expect tokio-modbus to recover from such situations, which is why I've encapsulated the `read_holding_registers` call in a `tokio::time::timeout` and drop the client session after every error/timeout.

However, tokio-modbus currently tries to recover from that and thereby enters an infinite loop without delay or even exponential backoff. My timeout kills that read operation after a few seconds without data, but in the meantime, it retries every millisecond and thereby causes heavy CPU load.

In particular, I'm talking about this loop:
https://github.com/slowtec/tokio-modbus/blob/b8bdb90453e0c6c7caf5e7eb0c0a7ee44a93f55a/src/codec/rtu.rs#L242-L263

`get_pdu_len` reads a zero byte here, therefore returns with an `Err` that is caught by the `or_else` branch. That branch calls `recover_on_error` to clear that zero byte, sets `retry` to `true`, and tries again immediately.

I'm open for any solution that adds a short delay here.
Either via a `retry_delay` parameter that is passed to `tokio::time::sleep` before every retry, or via a passed function that is called before every retry (allows the user to implement exponential backoff).

CC @flosse @uklotzde

uklotzde, 2023-04-17:
This edge case has probably not been considered back then. The loop should definitely terminate if no progress is made and recovery fails. I would prefer exiting the loop and pass control back to the client application which could handle the retry depending on the context.

uklotzde, 2023-04-17:
I remember that we needed this recovery loop to handle spurious failures from a device that was connected via a Serial-over-USB adapter. The code is from pre-async Tokio 0.3 were recovery from errors in async code was much more complicated.

Making the library code less stateful and opinionated should be the goal. If you see a chance to simplify the code by removing this poor-man's approach for recovery from errors then don't hesitate to do it.

ColinFinck, 2023-04-18:
> I remember that we needed this recovery loop to handle spurious failures from a device that was connected via a Serial-over-USB adapter.

Recovering from _spurious_ failures that actually happen in practice sounds like a good reason to keep this code instead of giving up immediately and passing control back to the caller.

Instead of adding a short delay, another option would be to define a maximum number of retries.
The `retry` boolean variable would be replaced by a counter that is incremented on every retry and reset to zero on every success. When the maximum number of retries is reached without success (e.g. 20), we could fail and pass back control to the caller. What do you think about that?

uklotzde, 2023-04-18:
Sure, a retry limit would also be reasonable. Ideally configurable, but a constant is probably sufficient for now.

PR #186 (2023-04-24): Max retries
As suggested in #184, I've fixed the infinite loop when reading from a broken Modbus RTU bus by introducing a maximum number of retries. When that number is reached, the code gives up and returns an error to the caller.

`MAX_RETRIES` is a constant that is set to 20 for now. Feel free to change that or make it configurable.

I had to edit other parts of the RTU code, which assumed that decoding can never fail. I took the opportunity to make them more readable and more consistent with the TCP code. Should get obvious when you check this PR commit by commit.

Let me know if this can  [... cut]
closed 2023-04-24 by uklotzde