slowtec/tokio-modbus #185: "Device or resource busy" in RTU server example

gigantino, 2023-04-17:
Hello!

I would like to say in advance that this is one of my first times working with the modbus protocol and, mostly, with Rust.

Recently I've been working with a PLC (Xenje XC3-24RT-E) and wanted to build a small interface for it. I managed to make it work almost on the first attempt using the first Node.js module I could find, but since I wanted to experiment with [Tauri](https://tauri.app) I thought I might as well try and experiment a bit with Rust.

So, I cloned the repo and ran the [rtu-server](https://github.com/slowtec/tokio-modbus/blob/main/examples/rtu-server.rs) example, but after building it gave me the following error:
```SH
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { kind: Unknown, description: "Device or resource busy" }', examples/rtu-server.rs:53:68
```
I then tried to check if the port was being used somewhere else, and I confirmed it wasn't by running the command `lsof /dev/ttyUSB0` ([lsof](https://linux.die.net/man/8/lsof)).

I would also like to note that in my Node.js code I'm using the following port settings, so maybe I'm doing something wrong by not changing the default example. 
```JSON
{
  "baudRate": 19200,
  "parity": "even",
  "dataBits": 8,
  "stopBits": 1
}
```

Thank you to anyone who's willing to help.

thienandangthanh, 2023-07-22:
I find out that it's used again in this line.
https://github.com/slowtec/tokio-modbus/blob/5fc09c4ef31eaf8d9fb55eccb84230e6cc429fd7/examples/rtu-server.rs#L53

And I still don't know how to make it work.

I think the examples and docs should be more easy to understand. Their purpose is to guide library user, not to confuse us.

thienandangthanh, 2023-07-22:
May we should use other device path, like `"/dev/ttyUSB1"` to create the builder for client_serial instead of using the same builder.

wheregmis, 2023-11-28:
You can create a virtual serial ports using socat.

and run 
`
socat -d -d pty,raw,echo=0 pty,raw,echo=0
`
You will get two socket 
`
2023/11/28 15:10:19 socat[66531] N PTY is /dev/ttys031
2023/11/28 15:10:19 socat[66531] N PTY is /dev/ttys033
2023/11/28 15:10:19 socat[66531] N starting data transfer loop with FDs [5,5] and [7,7]
`

Use one for the server and another for the client
`
    let server_builder = tokio_serial::new("/dev/ttys031", 19200);
    let client_builder = tokio_serial::new("/dev/ttys033", 19200);
`

Use these builder for server and client. It will work.

chadwagner1, 2024-08-20:
I had to run using sudo under Ubuntu to get arround this problem.

gigantino, 2024-08-20:
I don't have means to test what @chadwagner1 said as of right now, if it's that simple and someone can confirm we can probably close this issue and open a PR to document it better