slowtec/tokio-modbus #190: Generalize impl around `Server` usage

Halkcyon, 2023-05-19:
In previous versions of the crate, I could wrap my `Service` in an `Arc` to pass into `Server::serve`.  In the `0.8` release, I needed to create a new delegate type that basically accomplishes the same thing, i.e.,

```rs
struct MbDelegate(Arc<MbServer>);

impl Service for MbDelegate {
// ...

  fn call(&self, req: Self::Request) -> Self::Future { self.0.call(req) }
}
```

A suggestion to make this delegation unnecessary, create a blanket impl, such as:

```rs
impl<D> Service for D
where
  D: Deref,
  D::Target: Service + ?Sized
```

So that anything that derefs into a `Service` can be used.  As it stands, re-using a service (versus the example documentation where it appears a new service is created on every request), is kind of cumbersome and doesn't seem to make a lot of sense for a stateful server.

uklotzde, 2023-05-22:
Sound reasonable. Please submit a PR with the proposed changes.

What are the consequences for users of this crate? Is this a breaking change?

Halkcyon, 2023-05-22:
hey @uklotzde I wanted to open the issue for discussion before putting pen to paper.  This should not be a breaking change since users shouldn't be able to blanket impl a third-party trait on a different third-party type without some kind of wrapper already.

It's mostly an ergonomics change to create stateful servers as was possible in previous versions.

uklotzde, 2023-05-22:
Then go ahead ;)

Halkcyon, 2023-05-22:
I just realized this should read `Service` instead of `Server` in the titles, but can't update them.  The PR is open now @uklotzde

uklotzde, 2023-05-22:
> I just realized this should read `Service` instead of `Server` in the titles, but can't update them. The PR is open now @uklotzde

You should be able to edit titles and descriptions of your own issues and PRs that you created :thinking: I am not aware that this could be restricted for a repository.

PR #195 (2023-05-23): Generalize impl around `Service` usage
Resolves #190
closed 2023-05-23 by uklotzde