slowtec/tokio-modbus #191: Tokio module 'runtime' is private

ShayBox, 2023-05-21:
[image]

ShayBox, 2023-05-21:
This appears to be caused by this https://github.com/tokio-rs/tokio/blame/master/tokio/src/lib.rs#L536-L541

flosse, 2023-05-21:
... so it's not related to `tokio-modbus`, right?

ShayBox, 2023-05-21:
This is related to `tokio-modbus`, the error comes from it trying to access the `runtime` tokio module without the `cfg_rt!` (I don't know what that macro expands to, likely a cfg feature flag?)

flosse, 2023-05-21:
hm... I tried to compile some examples and could not reproduce this error. Could you provide some more info to reproduce it?

ShayBox, 2023-05-21:
`Cargo.toml`
```toml
[package]
name = "test"
version = "0.0.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
tokio-modbus = { version = "0.8.0", default-features = false, features = ["rtu-sync"] }
tokio-serial = "5.4.4"
```

`src/main.rs` [examples/rtu-client-sync.rs](https://github.com/slowtec/tokio-modbus/blob/main/examples/rtu-client-sync.rs)
```rust
// SPDX-FileCopyrightText: Copyright (c) 2017-2023 slowtec GmbH <post@slowtec.de>
// SPDX-License-Identifier: MIT OR Apache-2.0

//! Synchronous RTU client example

fn main() -> Result<(), Box<dyn std::error::Error>> {
    use tokio_modbus::prelude::*;

    let tty_path = "/dev/ttyUSB0";
    let slave = Slave(0x17);

    let builder = tokio_serial::new(tty_path, 19200);

    let mut ctx = sync::rtu::connect_slave(&builder, slave)?;
    println!("Reading a sensor value");
    let rsp = ctx.read_holding_registers(0x082B, 2)?;
    println!("Sensor value is: {rsp:?}");

    Ok(())
}
```

ShayBox, 2023-05-21:
It appears to be because of `default-features = false`, it's been a while since I touched my project and I just updated it, but [this comment](https://github.com/slowtec/tokio-modbus/blob/main/Cargo.toml#L73) implies it's still an intended use-case, probably one of the optional dependencies from a different feature is explicitly required.

flosse, 2023-05-21:
I'm so sorry, I'm still not able to produce the error, even if I disable the default features :thinking:
This is what works fine on my machine:
```sh
cargo check --no-default-features --features rtu-sync --example rtu-client-sync
```

ShayBox, 2023-05-21:
Yes that command works if I clone tokio-modbus and run it within, this only happens in a new project with tokio-modbus as a dependency

ShayBox, 2023-05-21:
Here's my test project if you like, it's just the above two files [Test.zip](https://github.com/slowtec/tokio-modbus/files/11525567/Test.zip)  
`$ cargo build`

flosse, 2023-05-21:
I see...you're right!

flosse, 2023-05-21:
Could you check if #192 solves the problem?

ShayBox, 2023-05-21:
That does fix it :+1: