slowtec/tokio-modbus #203: TCP Sync Client

gxcora, 2023-07-04:
Hi:
I'm still reasonably new to Rust. I tried to run your client example from here https://github.com/slowtec/tokio-modbus/blob/main/examples/tcp-client-sync.rs, but it gave me the error via vscode: `failed to resolve: use of undeclared crate or module sync` 
Any ideas?
Thanks
G
`

uklotzde, 2023-07-04:
You need to enable the required features manually: https://github.com/rust-lang/cargo/issues/4663

```
cargo run --example tcp-client-sync --features tcp-sync
```

closed 2023-07-04 by uklotzde

gxcora, 2023-07-04:
I did that, and it returns `error: none of the selected packages contains these features: tcp-sync`

uklotzde, 2023-07-04:
I tested it on the `main` branch and it works.

uklotzde, 2023-07-04:
I don't know anything about your private project. But it looks like you forgot to add the features to the dependency.

gxcora, 2023-07-05:
Thanks Uwe. That was my bad. I got it to build ok. I changed the example like this
```
fn main() {
    use tokio_modbus::prelude::*;

    let socket_addr = "10.2.16.171:502".parse().unwrap();
    let mut ctx = sync::tcp::connect(socket_addr).unwrap();
    let buff = ctx.read_input_registers(3, 1).unwrap();
    println!("Response is '{buff:?}'");
}
```

And it returns 
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
     Running `../tokio-modbus/target/debug/examples/tcp-client-sync`
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: Other, error: ExceptionResponse { function: 4, exception: IllegalDataAddress } }', examples/tcp-client-sync.rs:11:47
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Yet mbpoll is able to see the data:
```
mbpoll -m tcp -a 1 -t 4 -r 3 -c 1 -1 10.2.16.171
......

Protocol configuration: Modbus TCP
Slave configuration...: address = [1]
                        start reference = 3, count = 1
Communication.........: 10.2.16.171, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[3]: 	2
```

Any ideas?

uklotzde, 2023-07-05:
Check the tokio-modbus docs about 0/1-based addressing.

AJAnderson, 2024-09-20:
I'm having the same issue as the OP:

I have tried both:
[dependencies]
tokio-modbus = "*"

and 
[dependencies]
tokio-modbus = { version = "*", default-features = false, features = ["tcp"] }

in my .toml file.

My main.rs file in the src folder has the code from https://github.com/slowtec/tokio-modbus/blob/main/examples/tcp-client.rs

Can you help me resolve the confusion please?

AJAnderson, 2024-09-20:
Further: when I invoke cargo run --features tcp I get
"none of the selected packages contains these features: tcp"