slowtec/tokio-modbus #205: write_multiple_registers taking array as a param

jackstar12, 2023-07-21:
I am currently using this library (really like it) and was wondering why the `write_multiple_registers` function takes the input in form of an array? This is a bit of a pain and the data is turned `to_vec()` directly anyways. 

https://github.com/slowtec/tokio-modbus/blob/5fc09c4ef31eaf8d9fb55eccb84230e6cc429fd7/src/client/mod.rs#L289-L308

Is there any intention behind this?

uklotzde, 2023-07-21:
I don't remember the details. If you have an idea how to improve the API please submit a PR for discussion.

jackstar12, 2023-07-21:
> I don't remember the details. If you have an idea how to improve the API please submit a PR for discussion.

I thought of changing it to `Vec<Word>` because thats whats used internally aswell. Ill create a pr

uklotzde, 2023-07-22:
Using `Cow` might be beneficial here. This will cause a breaking API change and inconveniences for users. The latter could be mitigated by introducing helper methods. I am exploring such an extension.

PR #207 (2023-07-25): Request: Do not allocate a temporay `Vec` when writing multiple coils/registers
Fixes #205.

It turned out that the temporary allocations are not needed. Passing a borrowed slice is fine.

The traits didn't change. Only `Request` requires a lifetime parameter.

### TODO

- [x] Rebase after #206 has been merged.
closed 2023-07-25 by flosse