slowtec/tokio-modbus #210: Can not disconnect slave, need help

ileon, 2023-07-25:
async fn main() -> Result<()> {
    {
        let mut ctx1 = get_ctx(&get_default_sp_builder(), 2)?;
        let rsp1 = ctx1.read_holding_registers(1, 2).await?;
        println!("{rsp1:?}");
        ctx1.disconnect().await?;
    }

    let mut ctx2 = get_ctx(&get_default_sp_builder(), 1)?;
    let rsp2 = ctx2.read_holding_registers(1, 2).await?;
    println!("{rsp2:?}");
    ctx2.disconnect().await?;

    Ok(())
}

above is my code, the rsp1 is right, but failed at the ctx2 row
What is the right method? thanks!

uklotzde, 2023-07-25:
Context? Fails how?

ileon, 2023-07-27:
> Context? Fails how?

there are two contexts, the ctx1 works normal, then the ctx1.disconnect().await? is not take effect, so the ctx2 panic because access is denied.

uklotzde, 2023-07-27:
`get_ctx()` and `get_default_sp_builder()` are not provided by tokio-modbus and I cannot comment on this. Everything might happen behind the scenes.

Please re-open if you could provide a runnable or at least comprehensible example.

closed 2023-07-27 by uklotzde