slowtec/tokio-modbus #219: Patterns/best practice for reconnect-after-failure?

PeterGrace, 2023-09-03:
I'm utilizing tokio-modbus with tokio-retry to do exponential backoff on a tcp-based modbus connection.

This connection can be inconsistent and the tcp stream can fail.  When this happens, I get an indication in the form of an Err() that I can parse to figure out that we've lost the tcp stream entirely.

My question: What's the best practice on disconnecting and reconnecting to the same socketaddr/slave_id that was being used in the context prior to a disconnect?  I'm trying to do a ctx.disconnect() followed by creating a new ctx object with a fresh connection via tcp::connect_slave() call but I'm wondering if there's a better/more expected way to handle this scenario?

dr-kernel, 2023-11-28:
I would very much like to know this too.  
Currently I'm doing the same as Peter.

1787611685, 2024-12-16:
I also want to know how to gracefully determine that the connection has been lost and exit the loop reading to re-establish the connection when the Modbus TCP connection is lost. The reconnection mechanism I am currently using is the same as yours.