slowtec/tokio-modbus #252: tokio-modbus 0.12 entering unreachable code in any response error

gustavowd, 2024-03-26:
Hi,

Since tokio-modbus 0.12 i'm facing several application crashs. The reason is that in any returning error code the tokio-modbus enters an unreachable code. I'm communicating with a commercial equipament, which sometimes returns a invalid response. In tokio-modbus 0.11 a handle this erros, but now it always causes a application crash.

Mar 26 08:12:14 raspberrypi gateway.sh[29153]: thread 'main' panicked at /home/gateway/.cargo/registry/src/index.crates.io-1cd66030c949c28d/tokio-modbus-0.12.0/src/client/mod.rs:230:25:
Mar 26 08:12:14 raspberrypi gateway.sh[29153]: internal error: entered unreachable code: unexpected response code: 4 (request code: 3)
Mar 26 08:12:14 raspberrypi gateway.sh[29153]: note: Please report the issue at `https://github.com/slowtec/tokio-modbus/issues` with a minimal example reproducing this bug.
Mar 26 08:12:14 raspberrypi gateway.sh[29153]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

Thanks in advance,
Gustavo

uklotzde, 2024-03-26:
https://github.com/slowtec/tokio-modbus/pull/248#issuecomment-1955226882

Looks like some paths are marked as unreachable, although they could be reached if a server sends an invalid/unexpected response.

creberust, 2024-03-26:
Hi,

We can remove the `unreachable!` parts, but we need to add an `Error` variant, or return an `io::Error` ?

I think we can return an `std::io::Error` with `ErrorKind::InvalidData` in case of `Request`/`Response` mismatch. I don't how you handled it before ?

Would it be simple for you @gustavowd to handle the `Error` with this ?

uklotzde, 2024-03-26:
I'll prepare a proposal.

uklotzde, 2024-03-26:
Please review and test #254 thoroughly!

uklotzde, 2024-03-26:
Due to this serious panic issue I have decided to yank v0.12.0. Not recommended for production use.

gustavowd, 2024-03-26:
> Hi,
> 
> We can remove the `unreachable!` parts, but we need to add an `Error` variant, or return an `io::Error` ?
> 
> I think we can return an `std::io::Error` with `ErrorKind::InvalidData` in case of `Request`/`Response` mismatch. I don't know how you handled it before ?
> 
> Would it be simple for you @gustavowd to handle the `Error` with this ?

I beleive that @uklotzde proposal solves the issue. Yes, it is simple to handle the error with a returning error.

PR #254 (2024-03-27): Fix panic on mismatching request/response function codes
Fixes #252.
closed 2024-03-27 by flosse