slowtec/tokio-modbus #257: How to get hexadecimal modbus rtu message data

byte-voyager, 2024-03-29:
```
impl tokio_modbus::server::Service for Service {
    type Request = SlaveRequest<'static>;
    type Future = future::Ready<Result<Response, Exception>>;

    fn call(&self, req: Self::Request) -> Self::Future {
        match req.request {
            Request::ReadInputRegisters(_addr, cnt) => {
                let mut registers = vec![0; cnt.into()];
                registers[2] = 0x77;
                future::ready(Ok(Response::ReadInputRegisters(registers)))
            }
            Request::ReadHoldingRegisters(_, _) => {
                future::ready(Err(Exception::IllegalDataAddress))
            }
            _ => unimplemented!(),
        }
    }
}
```

It doesn't seem to get the full byte message in this example

uklotzde, 2024-03-29:
Is this a question or is this a bug report? Please provide a working, reproducible example.

byte-voyager, 2024-04-11:
> Is this a question or is this a bug report? Please provide a working, reproducible example.

 Hi, want to use for logging full messages to the database

closed 2025-07-10 by byte-voyager