slowtec/tokio-modbus #263: SECS/GEM SEMI?

han1548772930, 2024-07-12:
Hello, I would like to ask if this can be used for SECS/GEM SEMI?

jerrywrice, 2024-08-20:
If you're asking if Modbus is usable for accessing/controlling sensors and/or actuators from a running SECS/GEM interface library, then there is no inherent issue - assuming you use the Rust FFI or have a native Rust SECS/GEM library. If you're instead asking if a Modbus protocol port can be directly connected as a matching remote end-port to a SECS/GEM link, then the answer is no. Modbus and SECS/GEM are totally  incompatible network protocols.

han1548772930, 2024-08-21:
> If you're asking if Modbus is usable for accessing/controlling sensors and/or actuators from a running SECS/GEM interface library, then there is no inherent issue - assuming you use the Rust FFI or [...]

Thanks for your reply

closed 2024-08-21 by han1548772930